mantidproject/mantid #20052: Improve FindSXPeaks

AntonPiccardoSelg, 2017-07-20:
SXPeaks currently finds the strongest peaks per spectrum and returns a peak list. The background calculation is based on data in the current spectrum. We want to:

* add an absolute background for all spectra
* add a peak finding strategy which finds all peaks above a certain strategy
* (maybe handles the peak reduction better)

In addition the algorithm does not handle `WISH` and `ALF` data. This needs to be fixed.

samueljackson92, 2017-07-20:
Reading about the problems uncovered with FindPeaksMD + the fact that ConvertToMD for one WISH file takes ~10mins makes this a good idea. 

It would be good to know how long it takes to search for peaks in each spectrum when you have as many as WISH. If this is very slow you could maybe cut down the search space by doing something like:
 - Integrate all spectra
 - Arrange integrated pixels spatially in 2D (like we do with a cylindrical mapping in the instrument view)
 - Find peaks + a radius in this 2D space.
 - Only perform 1D peak search only along spectra that fall in this radius.

That may or may not be slower than a brute force search.

AntonPiccardoSelg, 2017-07-21:
@samueljackson92 Thanks for commenting. I have training today, but I will be back on Monday. Let's talk about some of your suggestions then.

PR #20104 (2017-08-16): Improve FindSXPeaks
Fixes #20052 

This PR enhances `FindSXPeaks` to find all peaks in a spectrum above a certain threshold. The old (and quicker) behaviour of finding only the strongest peak per spectrum can still be selected.

## To test:

Please find a pre-prepared sample data set here: smb://olympic/babylon5/Scratch/Anton/TEST_DATA_FIND_PEAKS_MD

You can use your own data set if you have one.

### Ensure that default runs

1. Load the file `wish_event_with_peaks.nxs` and make sure the workspace has the name `wish_event_with_peaks`
2. Open the `FindSXPeaks` dialog
2. Set `PeakFindingStrateg [... cut]
closed 2017-08-16 by ianbush