mantidproject/mantid #20070: ISIS Reflectometry: Remove progress bar animation effect when reduction paused

PranavBahuguna, 2017-07-24:
When reduction is paused in the interface, the progress bar stops updating as well. However it still shows an animation effect as though reduction is still occurring. This effect should be removed to make it more clear that no reduction is currently occurring.

PranavBahuguna, 2017-07-25:
In retrospect, this issue makes little sense as it only applies to Windows isn't trivial to implement anyway. I'm closing this for now unless it is deemed important enough in the future to be worth doing.

gemmaguest, 2017-09-07:
It looks like this is difficult and would not give a very ideal solution anyway. Since it is very low priority I don't think it is worth doing. @edwardb96, could you add a link to the information you found on this and then close this issue?

edwardb96, 2017-09-08:
This [stackoverflow thread](https://stackoverflow.com/questions/16556482/disable-progress-bar-animation-in-qt) shows one possible solution, however doing this would make the progress bar inconsistent with the native appearance, it also looks a bit windows 95.

Skimming over the qt documentation for the QProgressBar [widget](http://doc.qt.io/qt-4.8/qprogressbar.html) and [style options](http://doc.qt.io/qt-5/qml-qtquick-controls-styles-progressbarstyle.html) there doesn't appear to be a simple way of disabling this effect (probably because it's windows specific behaviour).

closed 2017-09-08 by edwardb96