mantidproject/mantid #20086: Multiple plot normalisation issue

gvardany, 2017-07-25:
When 2 spectra are plotted in separate windows, MantidPlot confuses the toggling of their normalisations (bin width/none).

### Expected behaviour
Normalise each spectrum individually in it's window.

### Actual behavior
They are mixed

### Steps to reproduce the behavior
```
ws1 = CreateSampleWorkspace()
ws2 = CloneWorkspace(ws1)
```
Plot the spectrum 1 of each of the 2 workspaces in separate windows.
On one of the windows toggle the normalisation option (to none).
Then click on the 2nd window, right click and you will see that the normalisation is shown as none, but in reality it is still bin-width normalised.
If you now toggle the normalisation on window 2, it will act on the spectrum in window 1.

### Platforms affected
ALL

mantid-roman, 2017-11-28:
I don't see this on Windows. Could it be mac only issue?

gvardany, 2017-11-28:
Yes I just realised that. Seems it's mac os that confuses the focus.

closed 2021-03-02 by stale[bot]