mantidproject/mantid #20088: ISIS Reflectometry: Allow setting multiple transmission runs for specific angles

PranavBahuguna, 2017-07-25:
One of problems instrument scientists have with the current interface is that the `Transmission run(s)` setting in the `Settings` tab simply applies any specified transmission runs to all runs in the processing table. It lacks any capability to specify different transmission runs for runs with different specific angles, so they are forced to input these transmission runs in the table directly.

The `Transmission run(s)` setting needs to be changed to allow the user to input up to 2 transmission runs per angle. One possible way of doing this is to let the user input a map of angles to the transmission runs that can later be parsed. For example:

```
{angle=0.2}
{angle=0.7, firstTranmissionRun=INTER00013463}
{angle=1.5, firstTranmissionRun=INTER00013463, secondTransmissionRun=INTER00013464}
```

In the `GenericDataProcessorPresenter`, we should parse this input to apply any specified transmission runs to all runs with that angle key.

_It is a good idea to talk to instrument scientists such as Max Skoda to clarify any other requirements before starting this issue._

gemmaguest, 2017-09-07:
Max's original suggestion was to do this as a table, which would probably be the most user-friendly way of inputting the data, although Pranav thought this may be difficult to implement. Max doesn't mind if we use an alternative input based on the existing single text box, however it needs to be easy for users to edit/read and easy for Mantid to parse. Users are familiar with python so perhaps something like a dictionary initialization syntax could be an option. We could also consider a multi-line text entry box, e.g.
```
0.7 : 12345, 12346
2.3 : 23456, 23457
```

Note that we usually have 2 angles, but there may be more (probably 4 at most).

gemmaguest, 2017-10-10:
Note that angle lookup should match to about 100th of a degree, i.e. if the table has an angle of 0.7 but the run has an angle of 0.69, then this should match.

gemmaguest, 2017-12-14:
I'm going to look at this next as part of #21167

gemmaguest, 2018-02-05:
Max mentioned that there is also a requirement on Crisp to specify lambdaMin and lambdaMax on a per-angle basis. He suggested adding additional columns to the transmission runs table to essentially reproduce the table on the Runs tab so that all properties can be specified on a per-angle basis (note however that lambdaMin and lambdaMax can only be specified via the Options column on the Runs tab - there is no dedicated column).

Mario would actually prefer to keep the autofill functionality. @edwardb96 suggested that this could be done using a filter on the Runs tab and then autofilling the filtered rows. This could be a neater approach (we need to add a filter anyway). However, the angle may not be populated in the table initially when a run is typed in manually. Perhaps this angle could be populated after the run number is entered, or when a filter is performed, but this could be slow because it means loading the runs.

martyngigg, 2018-03-06:
Closed after merging #21762 to release branch.

closed 2018-03-06 by martyngigg