mantidproject/mantid #20089: ISIS Reflectometry: Exclude transmission run from processing

PranavBahuguna, 2017-07-25:
When autoreduction is started, all runs including the transmission run are selecting for processing. When it attempts to process the transmission run however, it throws an error as it has an angle of zero (all valid runs must have an angle greater than zero).

This transmission run should probably be excluded from the list of runs to be processed. It would also be worth talking to the instrument scientists about any other runs (such as `Alignment check`) that should also be excluded from processing.

**To test:**

- Login to the ICAT catalog
- Open the ISIS Reflectometry interface, change instrument to `POLREF` and enter id `1520249`.
- While you can run autoreduce until you reach the last run to see the error, it is faster to simply search and transfer all runs and process the last row 'Transmission' - it will throw an error as it has an angle of zero.

gemmaguest, 2017-09-07:
Following discussion with Max, there doesn't seem to be a good way to identify which runs to exclude - they should be sensibly named but we can't necessarily rely on it. They don't actually care whether these runs get reduced as long as they don't cause errors which stop the autoreduction.

gemmaguest, 2018-06-18:
This will be fixed by PR #22443 which excludes runs with an invalid angle

PR #22443 (2018-06-21): Autoprocessing on reflectometry GUI
**Description of work**
This PR adds fully automatic processing of an investigation on the `ISIS Reflectometry` GUI.
- The `Autoprocess` button transfers all rows to the table and processes them.
- When all runs have been processed, a background processes is started to poll for new runs from ICat. If any are found, they are also transfered and processed.
- Some minor refactoring has been done to reduce complexity. The queues of items to process have been removed from `GenericDataProcessorPresenter` and instead the processed state of rows is used to skip rows [... cut]
closed 2018-06-21 by SimonHeybrock