mantidproject/mantid #20163: Parse the instrument tree in the base instrument

SimonHeybrock, 2017-08-03:
Parsing the instrument tree in the base instrument when it is being added to the IDS has two big advantages when loading more than one workspace with the same instrument:
- We avoid repeated parsing of the same tree.
- Data in instruments (`DetectorInfo` and `ComponentInfo`) will be shared between workspaces. Previously this was the case only for workspaces that were created from a parent workspaces, but not indenpendently created workspaces with the same instrument. Note that if there are parameters that modify the instrument sharing will still be (partially) broken.

SimonHeybrock, 2018-09-10:
@OwenArnold I believe we implemented this, correct?

OwenArnold, 2018-09-10:
@SimonHeybrock correct

closed 2018-09-10 by SimonHeybrock