mantidproject/mantid #20196: BFGS empty matrix for fit benchmark

AnthonyLim23, 2017-08-08:
in fit benchmarking running: GEM63437_focused_bank3.nxs with BFGS causes an error: 

IndexError: Matrix is empty.

This error seems to come from the fitting function that has been chosen.

Anders-Markvardsen, 2017-08-08:
Is this a bug only when the user selects minimizers = ['BFGS', 'Damped GaussNewton']  or do you also see it when you see all the minimizers

AnthonyLim23, 2017-08-08:
I have been talking to Roman about it. He says that it is BFGS and Conjugate gradient (Polak-Ribiere imp.)

He has also tracked the error down to void DerivMinimizer::initialize

AnthonyLim23, 2017-08-08:
Talking to Roman it the least squares file used the wrong command in one line it had `evalDeriv` instead of `evalHessian`

I have made the change to the code and I am going to run all of the system tests to make sure nothing as broken.

Anders-Markvardsen, 2017-08-08:
uummm odd that this only affected BFGS though.... let me know how you would like me (or someone else) to test this

AnthonyLim23, 2017-08-08:
the BFGS uses a Hessian matrix to decide the direction of the minimisation. I think the error meant that the Hessian was not constructed.

PR #20198 (2017-08-11): BFGS empty matrix
BFGS did not always work when called from python, this seemed to be because the Hessian matrix was not constructed. 

To test check that all of the minimizers work when called from python. The easiest way is to run the fit benchmarking with all minimizers on neutron data.

Fixes #20196

**Release Notes** 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mant [... cut]
closed 2017-08-11 by martyngigg