mantidproject/mantid #20206: Implement new algorithm for binning of event workspaces in (d, d_perp) space

mganeva, 2017-08-09:
For TOF powder diffraction in 2D required a binning algorithm, which will bin the input event workspace in (d, d_perp) space. Algorithm should be able to read the binning parameters from the text file.

Preliminary name for the algorithm is `BinEvents2D`.

More details on the multidimensional TOF powder diffraction data reduction in:
J. Appl. Cryst. (2015) 48, 1627-1636
J. Appl. Cryst. (2017) 50, 866-875.

tags: powder diffraction, TOF 

Please, assign me.

AndreiSavici, 2017-08-09:
The name is not even hinting that this is diffraction related. Maybe something like `Bin2DPowderDiffraction`. Also, I'm not sure you have to restrict yourself to events. Is there a reason not to be able to use workspace2Ds?

mganeva, 2017-08-09:
For the moment, algorithm works only with EventWorkspaces. Do you think it makes sence to extend it for workspace2D?

AndreiSavici, 2017-08-09:
There might be instruments that don't have events. You could make it work with events, and for now just open a new issue to make it work with histograms. To test it, just rebin the event workspace with a fine step, and the image should be similar (you will get some small effects due to the finite size of the histogram bin

mganeva, 2017-08-09:
Thanks for the idea. I will check whether I will manage to extend it for workspace2d. If not, I will open a new issue for that.

mganeva, 2017-08-11:
@AndreiSavici @peterfpeterson 
I've pushed now a code draft for the algorithm, which I've used to generate an image for the presentation on Mantid meeting in June. There is still a lot of code brush to be done, I will continue to work on it in 3 weeks, after I will come back from my holidays.

closed 2017-09-25 by AndreiSavici