mantidproject/mantid #20219: ISIS_Powder - Use AbsorptionCorrections for samples instead of MayersSampleCorrection

DavidFair, 2017-08-10:


martyngigg, 2017-08-10:
@DavidFair Could you provide a better description here including a reason why?

`MayersSampleCorrection` can just calculate absorption if you disable the MS option so why use two separate ones?

DavidFair, 2017-08-10:
@martyngigg I didn't add a description as we were just splitting this work out from another branch.
Currently ISIS_Powder has support for absorption corrections on arbitrary samples, this incorrectly uses Mayers Sample Correction for both Vanadium runs and all other materials.

Now we check if we are focusing a Vanadium. If so we use MayersSampleCorrections in the script otherwise we use AbsorptionCorrections.

martyngigg, 2017-08-10:
@DavidFair Thanks for the clarification.

The absorption part of `MayersSampleCorrection` should work for arbitrary samples. It's just the MS correction part that only applies to isotropic scatterers.

PR #20223 (2017-08-22): ISISPowder use absorption correction on samples
**Description of work.**

Previously ISIS_Powder did not support absorption correction on non-vanadium samples. We now check whether a sample is vanadium or not. In both cases we use Mayers' Sample Correction, but in the case of a non-vanadium sample, we explicitly disable correction for multiple scattering (possibly to be added at some point in the future).

**To test:**

Run a focus on some powder data, first with the `do_absorb_corrections` flag set to `False`, then `True`. Plotting the resultant spectra on top of each other should show distin [... cut]
closed 2017-08-22 by peterfpeterson