mantidproject/mantid #20235: Multiple Scattering: Usage Improvments

martyngigg, 2017-08-11:
James has done some good work improving the documentation around our current set of multiple scattering corrections - see #20170 #20145.

We now need to improve how these algorithms can be used. In contrast to the other absorption corrections the two MS algorithms calculate & correct for absorption and multiple scattering rather than just calculating the correction factors and applying them separately.

Tasks to do:
* Create a `CalculateAbsorptionAndMultipleScatteringCylinder` algorithm: this will have the option to choose either `Carpenter` or `Mayers` for the method. It will output separate workspaces with the correction factors for absorption and multiple scattering.
* The current algorithms `MultipleScatteringCylinderAbsorption` and `MayersSampleCorrection` will use the new algorithm calculate the correction and then apply it to the data as before.
* Look into extending the carpenter correction for out-of-plane scattering

cc @peterfpeterson @Anders-Markvardsen @AndreiSavici

jamesphysics, 2017-08-14:
- D.F.R. Mildner, J.M. Carpenter, and C.A. Pelizzari Generalized Attenuation Correction Factor for Scattering from Cylindrical Targets http://dx.doi.org/10.1063/1.1686687 explains how to generalise the attenuation correction for out-of-plane scattering

- D.F.R. Mildner and J.M.Carpenter Improvements to the Chebyshev Expansion of Attenuation Correction Factors for Cylindrical Samples http://dx.doi.org/10.1107/S0021889890005258 discusses how the accuracy of the technique may be quantified and improved

jamesphysics, 2017-08-14:
To get a crude idea of how well both algorithms calculate the attenuation correction factor I follow on from an idea by @DavidFair and corrected some vanadium data using both algorithms in comparison with the CylinderAbsorption http://docs.mantidproject.org/nightly/algorithms/CylinderAbsorption-v1.html which is believed to be the most accurate and acts as a guideline. 

I had to make a minor modification to the MultipleScatteringCylinderAbsorption (MSCA) code just to allow the multiple scattering correction to be switched off so we could just look purely at the absorption corrections in each case.

```
# USER SETUP

# Note this script takes 5-15 minutes for multiple scattering corrections
# Radius is in cm
cylinder_radius = 0.4

# Details for MayersSampleCorrection
mayers_cylinder_height = 4.0
mayers_position = [0., 0., 0.]

#----------------------------------------------------------------------------------------------------------------------#

# Load vanadium
Load(Filename= 'POL95598.nxs', OutputWorkspace='POL95598')
NormaliseByCurrent(InputWorkspace='POL95598', OutputWorkspace='POL95598')
geometry_json = {'Shape': 'Cylinder', 'Height': mayers_cylinder_height, 'Radius': cylinder_radius, 'Center': mayers_position}
material_json = {'ChemicalFormula': 'V'}
SetSample(InputWorkspace='POL95598', Geometry=geometry_json, Material=material_json)
# TODO reset this back!!!
# Load background count
Load(Filename='POL95597.nxs', OutputWorkspace='POL95597')
NormaliseByCurrent(InputWorkspace='POL95597', OutputWorkspace='POL95597')
# subtract the background from the vanadium
Minus(LHSWorkspace='POL95598', RHSWorkspace='POL95597', OutputWorkspace='POL95598')
# get rid of 'negative' counts at very low TOF
CropWorkspace(InputWorkspace='POL95598', OutputWorkspace='POL95598', XMin=750, XMax=20000)
AlignDetectors(InputWorkspace='POL95598', OutputWorkspace='POL95598_reference', CalibrationFile='C:\Users\ezz64382\Desktop\MultipleScattering\InvestigationUsingPOLARISdata\cycle_16_3_silicon_all_spectra.cal')
# the first 55 detectors are unused detectors or monitors on Polaris
MaskDetectors(Workspace='POL95598_reference', SpectraList='1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55')

#CylinderAbsorption corrections
ConvertUnits(InputWorkspace="POL95598_reference", OutputWorkspace="POL95598_CA", Target='Wavelength')
ca_ws_abs_factors = CylinderAbsorption(InputWorkspace="POL95598_CA", CylinderSampleHeight=mayers_cylinder_height, CylinderSampleRadius=cylinder_radius)
POL95598_CA = Divide(LHSWorkspace="POL95598_CA", RHSWorkspace=ca_ws_abs_factors)

#Added flag to MultipleScatteringCylinderAbsorption to disbale multiple scattering corrections
ConvertUnits(InputWorkspace="POL95598_reference", OutputWorkspace="POL95598_MSCA", Target='Wavelength')
MultipleScatteringCylinderAbsorption(InputWorkspace="POL95598_MSCA", OutputWorkspace="POL95598_MSCA", MultipleScattering=False)

#Mayer's correction
ConvertUnits(InputWorkspace="POL95598_reference", OutputWorkspace="POL95598_Mayers", Target='TOF')
MayersSampleCorrection(InputWorkspace="POL95598_Mayers", MultipleScattering=False, OutputWorkspace="POL95598_Mayers")

# to dSpacing + focussing
focused_list = []
division_list = []
for ws in ["POL95598_reference","POL95598_MSCA", "POL95598_Mayers", "POL95598_CA"]:
    loop_ws = ConvertUnits(InputWorkspace=ws, OutputWorkspace=ws, Target='dSpacing')
    focused_list.append(DiffractionFocussing(InputWorkspace=ws, OutputWorkspace=ws, GroupingFileName='C:\Users\ezz64382\Desktop\MultipleScattering\InvestigationUsingPOLARISdata\Master_copy_of_grouping_file_with_essential_masks.cal'))

division_list.append(mtd['POL95598_CA']/mtd['POL95598_Mayers'])
division_list.append(mtd['POL95598_CA']/mtd['POL95598_MSCA'])
    
plot(focused_list, 3)

plot(division_list, 3)
```
This code outputs:

[image]

where the blue curve is the reference, the green and black curves are respectively the Mayers and Cylinder Absorption corrected data which appear to be in greater agreement than the MSCA correction. 

[image]

In the second graph I have divided each correction by the Cylinder Absorption correction. In order to verify the accuracy of these algorithms we would hope to see a flat line horizontal line close to unity which is roughly what we see for the Mayers correction.

For the MSCA this is not quite the case although promisingly, the relationship still appears to be linear.
It would be interesting to see a future comparison once this algorithm has been extended to handle out of plane scattering especially since it appears to remain stable over a larger interval of d-spacing than the Mayers correction (which is less smooth than the MSCA line and starts to really diverge from our "true" correction above 3 Angstroms).

DavidFair, 2017-08-14:
@jamesphysics - I've tided some hardcoded strings out of the script you posted so it should be more user friendly if other people tagged in this issue want to run your script on their machine. I also fixed a minor bug in the diffraction focussing loop (this will not affect your results).

Its interesting to note how Cylinder Absorption matches closely to Mayers. Was the Multiple Scattering Cylinder run with the multiple scattering component off (i.e. only comparing the absorption component between the 3 approaches)?

------

```
# USER SETUP

# Note this script takes 5-15 minutes for multiple scattering corrections
# Radius is in cm
cylinder_radius = 0.4

# Details for MayersSampleCorrection
mayers_cylinder_height = 4.0
mayers_position = [0., 0., 0.]

# Essential files required : - 
# Offset file:
cal_file_path = r"C:\Users\ezz64382\Desktop\MultipleScattering\InvestigationUsingPOLARISdata\cycle_16_3_silicon_all_spectra.cal"
# Grouping file:
grouping_file_path = r"C:\Users\ezz64382\Desktop\MultipleScattering\InvestigationUsingPOLARISdata\Master_copy_of_grouping_file_with_essential_masks.cal"

#----------------------------------------------------------------------------------------------------------------------#

# Load vanadium
Load(Filename= 'POL95598.nxs', OutputWorkspace='POL95598')
NormaliseByCurrent(InputWorkspace='POL95598', OutputWorkspace='POL95598')
geometry_json = {'Shape': 'Cylinder', 'Height': mayers_cylinder_height, 'Radius': cylinder_radius, 'Center': mayers_position}
material_json = {'ChemicalFormula': 'V'}
SetSample(InputWorkspace='POL95598', Geometry=geometry_json, Material=material_json)

# Load background count
Load(Filename='POL95597.nxs', OutputWorkspace='POL95597')
NormaliseByCurrent(InputWorkspace='POL95597', OutputWorkspace='POL95597')
# subtract the background from the vanadium
Minus(LHSWorkspace='POL95598', RHSWorkspace='POL95597', OutputWorkspace='POL95598')
# get rid of 'negative' counts at very low TOF
CropWorkspace(InputWorkspace='POL95598', OutputWorkspace='POL95598', XMin=750, XMax=20000)
AlignDetectors(InputWorkspace='POL95598', OutputWorkspace='POL95598_reference', CalibrationFile=cal_file_path)

# The first 55 detectors are unused detectors or monitors on Polaris
MaskDetectors(Workspace='POL95598_reference', SpectraList='1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55')

# CylinderAbsorption corrections
ConvertUnits(InputWorkspace="POL95598_reference", OutputWorkspace="POL95598_CA", Target='Wavelength')
ca_ws_abs_factors = CylinderAbsorption(InputWorkspace="POL95598_CA", CylinderSampleHeight=mayers_cylinder_height, CylinderSampleRadius=cylinder_radius)
POL95598_CA = Divide(LHSWorkspace="POL95598_CA", RHSWorkspace=ca_ws_abs_factors)

#Added flag to MultipleScatteringCylinderAbsorption to disable multiple scattering corrections
ConvertUnits(InputWorkspace="POL95598_reference", OutputWorkspace="POL95598_MSCA", Target='Wavelength')
MultipleScatteringCylinderAbsorption(InputWorkspace="POL95598_MSCA", OutputWorkspace="POL95598_MSCA", MultipleScattering=False)

#Mayer's correction
ConvertUnits(InputWorkspace="POL95598_reference", OutputWorkspace="POL95598_Mayers", Target='TOF')
MayersSampleCorrection(InputWorkspace="POL95598_Mayers", MultipleScattering=False, OutputWorkspace="POL95598_Mayers")

# to dSpacing + focussing
focused_list = []
division_list = []
for ws in ["POL95598_reference","POL95598_MSCA", "POL95598_Mayers", "POL95598_CA"]:
    loop_ws = ConvertUnits(InputWorkspace=ws, OutputWorkspace=ws, Target='dSpacing')
    focused_list.append(DiffractionFocussing(InputWorkspace=loop_ws, OutputWorkspace=loop_ws, GroupingFileName=grouping_file_path))

division_list.append(mtd['POL95598_CA']/mtd['POL95598_Mayers'])
division_list.append(mtd['POL95598_CA']/mtd['POL95598_MSCA'])
    
plot(focused_list, 3)

plot(division_list, 3)
```

jamesphysics, 2017-08-14:
@DavidFair thankyou and yes, MultipleScatteringCylinderAbsorption.cpp (and header) was modified to add an extra boolean input called 'MultipleScattering' which then is checked in the following section of code to decide whether to calculate deltp (the multiple scattering correction) or to leave it as zero and just include the attenuation component AttFac
```
double calculate_msa_factor(const double radius, const double Q2,
                            const double sigsct, const vector<double> &Z,
                            const double wavelength,const bool mscat) {

  const double sigabs = Q2 * wavelength;
  const double sigir = (sigabs + sigsct) * radius;
  const double sigsr = sigir;
  double deltp = 0.0;

  if (mscat) {
	  const double delta = COEFF4 * sigir + COEFF5 * sigir * sigir;
	  deltp = (delta * sigsct) / (sigsct + sigabs);
  }

  double temp = AttFac(sigir, sigsr, Z);
  return ((1.0 - deltp) / temp);
}
```

jamesphysics, 2017-08-15:
I didn't anticipate clang-format disliking the changes but none the less with the comments it should now be slightly easier to read with the aid of the original paper

jamesphysics, 2017-08-22:
[Mayers_MSCA.pdf](https://github.com/mantidproject/mantid/files/1242574/Mayers_MSCA.pdf)
Attached is a preview of a report I am writing on my investigation.

Some of what has been done above is repeated as well as some additional analysis. To summarise, the multiple scattering corrections in the MSCA and Mayers are closely in agreement however the MSCA appears to be deficient in its method of working out the absorption corrections.

As a result, Mayers sample correction is probably then the better of the two methods.

jamesphysics, 2017-09-05:
Multiple scattering cylinder absorption and Mayer sample correction currently get sample information in two different ways:
Mayers gets sample information from the input work space whereas Multiple scattering cylinder absorption takes the relevant information as arguments when it is called and if they are not given then as default the values are those of vanadium. This is fairly sensible however makes little sense in the case of the CylinderSampleRadius which is defaulted to 0.3175. This can (and has) led to some confusion because setting the workspace samples radius using the SetSample algorithm does not guarantee that Multiple scattering cylinder absorption will use the correct radius.

MSCA should probably work in a similar way to Mayers (where the sample geometry has been set already and is retrieved automatically) to bring the two algorithms into line although at the very least, CylinderSampleRadius should be made a Mandatory value to avoid confusion.

peterfpeterson, 2017-09-05:
We should have a little bit of thinking more on this since we want to "properly" account for containers and sample environment as well with a single approach that applies to multiple scattering and absorption.

peterfpeterson, 2017-09-11:
@jamesphysics put together a report [here](https://github.com/mantidproject/documents/blob/master/Presentations/Multiple%20Scattering%20Presentation/Investigating%20multiple%20scattering%20with%20McStas.docx).

peterfpeterson, 2017-09-11:
The "other" bit of FORTRAN that @jamesphysics found is in this scan https://www.isis.stfc.ac.uk/Pages/discus-manual6827.pdf

jamesphysics, 2017-09-11:
The FORTRAN has been typed out and is stored here. A bit more helpful than a scanned copy
https://github.com/mantidproject/3rdpartysources/tree/master/Fortran/Indirect/AbsCorrection

closed 2021-03-02 by stale[bot]