mantidproject/mantid #20250: Enforce the associative property for ProductFunction

jmborr, 2017-08-15:
Given functions `f`, `g`, and `h`, enforce that the three expressions `(f * g) * h`,  `f * (g * h)`, and `f * g * h` result in the same `ProductFunction` object.

jmborr, 2017-08-15:
@mantid-roman @KarlPalmen I have opened this issue to start addressing the issues that promted `castToCompositeFunction`

KarlPalmen, 2017-08-16:
Issue #19010 currently enforces the associative of a pure product function (one without any composite functions as members that are not also pure product functions), by means ``castToCompositeFunction``.
This was needed because ``ProductFunction`` does not accesses any product function it contains as an IFunction rather than a Composite Function.

jmborr, 2018-10-15:
this is being dealt with through the python API

closed 2018-10-15 by jmborr