mantidproject/mantid #20266: Expensive Check Component::hasComponentInfo

OwenArnold, 2017-08-17:
`Component::hasComponentInfo` is an important part of the compatibility mode we have in the transition between `Instrument 1.0` and `Instrument 2.0`. 

At the time of writing the implementation looks like this:

```
bool Component::hasComponentInfo() const {
  const IComponent *root = m_base;
  while (auto parent = root->getBareParent())
    root = parent;
  auto instrument = dynamic_cast<const Instrument *>(root);
  return m_map->hasComponentInfo(instrument);
}
```
see [here](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument/Component.cpp#L310-L316)

Which seems like an awfully expensive way of establishing whether we can forward to a `ComponentInfo` or not as it involves a **n** virtual function calls to `getBareParent()` in a loop, as well as a final `dynamic_cast`. The purpose of this seems to be to verify root `Component` of the present `Component` matches the instrument held by the `ParameterMap` instance held by the present `Component`.. before then yielding the `ComponentInfo` from the held `ParameterMap`.

Since we are now starting to use this check in a lot of [places](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument/Component.cpp#L338), it would be nice to simplify this somehow. Understandably we are still going to have to go via the `ParameterMap` but is the instrument check really necessary?

OwenArnold, 2017-08-17:
Looking at profiler outputs from this:

```cpp
//Instrumentation method I added to ExperimentInfoTestPerformance
  void test_temporary(){
    ExperimentInfo expInfo;
    expInfo.setInstrument(m_provisionedInstrument);
    auto source = expInfo.getInstrument()->getSource();
    CALLGRIND_START_INSTRUMENTATION;
    source->getPos().X();
    CALLGRIND_STOP_INSTRUMENTATION;
   }
```
of `Component::getPos()`  100%

---------------
`Component::index()` : ~52%
`Component::hasComponentInfo()` : ~15%
`API::ComponentInfo::position()` : ~33%

of greater interest is that of `API::ComponentInfo::position()` only ~0.4 of the time spent in that method is on `Beamline::ComponentInfo::position()`

SimonHeybrock, 2018-04-19:
@OwenArnold I do not understand where this is an issue. Where is this used frequently?

closed 2018-04-19 by OwenArnold