mantidproject/mantid #20369: Allow multiple event datasets to be loaded with LoadMcStas

Anders-Markvardsen, 2017-09-01:
LoadMcStas can read a mcstas output containing multiple event datasets, however it currently only load one of these. 

Fix this, update document for LoadMcStas accordingly (where relevant). 

Optionally (and if not this is done as separate issue). And small .h5 file from McStas simulation not larger than mcstas_event_hist.h5 with contains multiple event datasets and test (unit or usage example) which test against it.

jamesphysics, 2017-09-01:
Downloaded the user data. There is no file in the zipped folder called mcstas_event_hist.h5

Anders-Markvardsen, 2017-09-05:
mcstas_event_hist.h5 is in https://github.com/mantidproject/mantid/tree/master/Testing/Data/DocTest. @martyngigg my understanding was that this meant this file would automatically be zipped into the user data?

jamesphysics, 2017-09-05:
Attached is a zip containing an instrument file for an Incoherent source and three identical detectors set to measure, all, single and multiple scattering events respectively.
There is also a file containing the output data from each of the detectors when the instrument file was compiled and ran.
When loaded into Mantid using LoadMcstas the required outcome is that three separate workspaces are created for each of the data sets.

After putting a break point into the LoadMcstas code at line 119 (https://github.com/mantidproject/mantid/blob/master/Framework/DataHandling/src/LoadMcStas.cpp#L119) and 131 and trying to load the attached file, it appears that the three data sets are loaded three times, being overwritten each time until finally the last data set is the one actually output to the workspace. I'm not very familiar with loading and manipulating HDF files so I'm not sure exactly how to fix this problem. @michaeljturner may know more about this, I will ask him when he is back in the office.

[example.zip](https://github.com/mantidproject/mantid/files/1276883/example.zip)

michaeljturner, 2017-09-06:
Have edited LoadMcStas.cpp to correct this, and have sent it to @jamesphysics, since I'm on a different repo & branch and it would take too long to recompile with mantidproject/mantid to open a PR myself.

Anders-Markvardsen, 2017-09-26:
@trnielsen and @willend note this issue. Note it should not brake backward compatibility with existing mcstas-mantid user manual/instructions - but let know if you disagree

Anders-Markvardsen, 2017-11-28:
REMEMBER TO DELETE BRANCH 20369_fix_load_mcstas AFTER THIS ISSUE HAS BEEN FIXED

Anders-Markvardsen, 2017-11-28:
that is the branch 20369_fix_load_mcstas used for #20392

Anders-Markvardsen, 2017-12-08:
Attached in the zip is a McStas Nexus output from a vanadium sample, which contains data entries single_list, multi_list (means all event scattered more than once in this case) and all_list. For this issue, the LoadMcStas change should not make assumptions about what each of this event data entries are. It should simply return exactly the same as before, where EventData is the sum of all event entries, but in addition at the end of the workspace group also a workspace for each individual event Nexus entry. Note the dataset in the zip happens to contain no histogram data:

[mccode.zip](https://github.com/mantidproject/mantid/files/1542682/mccode.zip)

ElliotAOram, 2017-12-11:
@Anders-Markvardsen when I run `LoadMcStas` on the file in the zip, I get a warning in the Results log that states `Could not find the instrument description in the Nexus file:C:/Users/.../mccode.h5 Ignore evntdata from data file` and an **empty** workspace group is added to the ADS. Have I missed something?

Anders-Markvardsen, 2017-12-11:
The attached file may not work with the existing code it seems. Just looked inside it with HDFView and it does seem to contain an instrument: entry1->instrument->instrument_xml->data. My suggesting is that you work with the existing Usage Example mcstas file, which from memory I believe have two event-dataset and get these returned individually. Writing this, if the mcstas nexus only contain one event dataset then the behaviour should say as it is

PR #21423 (2018-01-15): LoadMcStas single, multiple and sum of scattering output
Previously the `LoadMcStas` algorithm only returned the sum of single and multiple scattering eventdata. 
This change means the algorithm now outputs both the single and multiple scattering data as well as the sum of the two. 
 
**To test:**

Run the LoadMcStas algorithm
*Note: The required file for the below is in the DocTest data directory*
```
test = LoadMcStas('mcstas_event_hist.h5')
```
* Observe that there are now 3 `EventData` workspaces: `EventData_test, EventData_test_1, EventData_test_2`
* `EventData_test_1` and `_2` should sum [... cut]
closed 2018-01-15 by martyngigg