mantidproject/mantid #20394: Interfaces working with python3

peterfpeterson, 2017-09-06:
This is an umbrella issue for fixing Interfaces with python3. The following interfaces don't work for a variety of reasons. Maybe not all of them are python3 related.

`Diffraction` -> `Engineering Diffraction` freezes mantid after logging
```
ApplicationWindow-[Information] Default facility 'SNS', instrument 'POWGEN'
FileFinder-[Information] found path = .../build/mantid/ExternalData/Testing/Data/UnitTest/ENGINX00241391.nxs
```

`Direct` -> `Convert to Energy (Direct)` logs a syntax error that is hiding in [Homer.cpp](https://github.com/mantidproject/mantid/blob/81b9c3d53435f150ae0764d6475778ba24fd5a12/qt/scientific_interfaces/General/Homer.cpp#L731)
```
LoadEmptyInstrument started
Algorithm: LoadEmptyInstrument v1
Execution Date: 2017-Sep-06 14:59:20
Execution Duration: -1 seconds
Parameters:
  Name: Filename, Value: /home/pf9/code/ma ... IA_Definition.xml, Default?: No, Direction: Input
  Name: InstrumentName, Value: , Default?: Yes, Direction: Input
  Name: OutputWorkspace, Value: __empty_SEQUOIA, Default?: No, Direction: Output
  Name: DetectorValue, Value: 1, Default?: Yes, Direction: Input
  Name: MonitorValue, Value: 2, Default?: Yes, Direction: Input
  Name: MakeEventWorkspace, Value: 0, Default?: Yes, Direction: Input
Loading geometry cache from /home/pf9/.mantid/instrument/geometryCache/SEQUOIA4dee7a163b7074e4873d9e47275b96873a474f7a.vtp
Parsing from XML file: /home/pf9/code/mantid/instrument/SEQUOIA_Parameters.xml
LoadEmptyInstrument successful, Duration 0.70 seconds
File "<Interface>", line 7
    print '{0}\n{1}\n{2}\n{3}\n{4}\n{5}\n'.format(int_range[0],int_range[1],prop_man.van_mass,bkg_rane[0],bkg_rane[1],str(prop_man.check_background))
                                         ^
SyntaxError: invalid syntax
```

`Muon` -> `Muon Analysis` crashes mantidplot after reporting that SNS is not a supported facility.
```
ApplicationWindow-[Information] Default facility 'SNS', instrument 'POWGEN'
Fatal Python error: This thread state must be current when releasing

Thread 0x00007f319d39e700 (most recent call first):
  File "<Interface>", line 2 in <module>

Thread 0x00007f319f596700 (most recent call first):
  File "/usr/lib64/python3.6/site-packages/zmq/utils/garbage.py", line 47 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f319fdd7700 (most recent call first):
  File "/usr/lib64/python3.6/site-packages/zmq/sugar/poll.py", line 99 in poll
  File "/usr/lib64/python3.6/site-packages/zmq/eventloop/ioloop.py", line 122 in poll
  File "/usr/lib64/python3.6/site-packages/tornado/ioloop.py", line 861 in start
  File "/usr/lib64/python3.6/site-packages/zmq/eventloop/ioloop.py", line 177 in start
  File "/usr/lib/python3.6/site-packages/ipykernel/iostream.py", line 72 in _thread_main
  File "/usr/lib64/python3.6/threading.py", line 864 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f31a182c700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 295 in wait
  File "/usr/lib64/python3.6/threading.py", line 551 in wait
  File "/usr/lib/python3.6/site-packages/IPython/core/history.py", line 834 in run
  File "/usr/lib/python3.6/site-packages/IPython/core/history.py", line 60 in needs_sqlite
  File "<decorator-gen-23>", line 2 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f3210a6b940 (most recent call first):
bin/launch_mantidplot.sh: line 49:  6999 Aborted                 (core dumped) LD_PRELOAD=${LOCAL_PRELOAD} TCMALLOC_RELEASE_RATE=${TCM_RELEASE} TCMALLOC_LARGE_ALLOC_REPORT_THRESHOLD=${TCM_REPORT} LD_LIBRARY_PATH=${LOCAL_LDPATH} QT_API=pyqt $VGLRUN$INSTALLDIR/MantidPlot $*
```

`Reflectometry` -> `ISIS Reflectometry OLD` reports
```
TypeError: addItems(self, Sequence[str]): argument 1 has unexpected type 'dict_keys'
  at line 9 in '<Interface>'
  caused by line 257 in '.../mantid/scripts/Interface/ui/reflectometer/refl_gui.py'
```

`SANS` -> `ISIS SANS v2 (experimental)` 
```
ValueError: The listener ist not of type SettingsDiagnosticTabListener but rather <class 'sans.gui_logic.presenter.settings_diagnostic_presenter.SettingsDiagnosticPresenter.ConcreteSettingsDiagnosticTabListener'>
  at line 25 in '<Interface>'
  caused by line 56 in '.../mantid/scripts/SANS/sans/gui_logic/presenter/main_presenter.py'
  caused by line 136 in '.../mantid/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py'
  caused by line 69 in '.../mantid/scripts/SANS/sans/gui_logic/presenter/settings_diagnostic_presenter.py'
  caused by line 57 in '.../mantid/scripts/Interface/ui/sans_isis/settings_diagnostic_tab.py'
```

peterfpeterson, 2017-09-06:
As a start to the issue with `Convert to Energy (Direct)` here is a [patch](https://github.com/mantidproject/mantid/files/1281758/homer.txt).

ElliotAOram, 2017-09-11:
The above patch works, however on Mantid with either version of python (2.7 or 3). This interface produces an error message box and error in the results log when loading with SNS POWERGEN

ElliotAOram, 2017-09-11:
I have been unable to repeat the freeze behaviour seen in the `Engineering Diffraction` interface

ElliotAOram, 2017-09-11:
It appears that running `.keys()` on a list in python 3 produces an object of type `dict_keys`, in python 2.7 this is a `str list`. The `2to3` tool doesn't pick this up so we should keep an eye out for this else where

ElliotAOram, 2017-09-12:
The ISIS SANS v2 interface was managing to import the same module twice and hence when using `isinstance` is to validate the listeners class, the Class id did not match as it had been re-imported.
The importing took place in the ui code for the data_processor_window.

closed 2017-09-15 by rosswhitfield