mantidproject/mantid #20421: Engineering Diffraction Analysis Interface - Initial dialog box

ElliotAOram, 2017-09-11:
Upon loading the Engineering diffraction interface (Interfaces > Diffraction > Engineering Diffraction)
The user is presented with the following dialogue box:
[image]
This dialogue can not be closed and the instruction do not inform the user that in order to use the interface (and close the dialogue box) you need to add an RB number to the input field.
A discussion should be had about how to make this clearer to the user. Options to consider are:
* Tooltip on input field
* Entering the RB number into the dialogue box
* Using the red `*` notation for required/incorrect parameter that is used in algorithms

josephframsay, 2017-10-31:
Addressed in PR #20993 (issue #20979). Possibly can be closed?

ElliotAOram, 2017-11-02:
@josephframsay agreed they are duplicate

closed 2017-11-02 by ElliotAOram