mantidproject/mantid #20440: Failing SystemTests.ISISDirectInelastic MAPS/MARI failing for python 3

rosswhitfield, 2017-09-12:
There are 3 remaining failing systemtests for python 3, see [here](http://builds.mantidproject.org/job/master_systemtests-ubuntu-16.04-python3/202/testReport/)
```
SystemTests.ISISDirectInelastic.MAPSDgreduceReduction
SystemTests.ISISDirectInelastic.MARIReductionAutoEi
SystemTests.ISISDirectInelastic.MARIReductionMonSeparate
```
They all fail with `Workspace ??? not equal to its reference file`

ElliotAOram, 2017-09-19:
Note: The above commit resolves issues for both `MARIReductionAutoEi` and `MARIReductionMonSeparate`

ElliotAOram, 2017-09-20:
The values for `Ei` in the sample logs of the reference files matched the python 2.7 version of the value returned from `str(ei)`. When investigating further, giving `format` a higher precision (up to 15 decimal places) it suggested that the `Ei` in the reference files was not correct past 9dp. In the example of the `MAPSDgreduceReduction` test, the `Ei` value in each file was:

| File | Ei value |
| :-: | :-: | 
| Reference file | 152.60479834399999 |
| Test output using `str()` | 152.60479834399999 |
| Test output using `format()` | 152.60479834443578 |

As the values are not accurate, the reference files for all 3 test cases have been updated to match the new `format()` output.

PR #20527 (2017-09-21): ISISDirectInelastic failing system tests for Python 3
The way the `Ei` value is formatted in the SampleLogs has been changed from using the `str()` function (which has a differing precision between python2.7 and python3.x) to the `format()` function which  explicitly quotes the precision.

The reference file were also found to not contain the correct precision for `Ei` so they have also been updated (See #20440 for more details on ref file update). @abuts would you be able to take a look at the change to the reference files and check that you are happy with this?

**To test:**

Ensure the syst [... cut]
closed 2017-09-21 by martyngigg