mantidproject/mantid #20443: CreateWorkspace excessively slow with larger datasets

MikeHart85, 2017-09-12:
### Summary

The execution time of `CreateWorkspace` quickly becomes unreasonably slow as you increase the size of DataX/DataY/DataE. It is orders of magnitude slower than using `WorkspaceFactory.create`. 

`AlgorithmHistory` is a major reason for this, but there appears to be more overhead elsewhere as well.

### Details

Two test functions to compare performance:
```python
def time_CreateWorkspace(nbins, nspec):
    import numpy
    xdata = numpy.arange(float(nbins + 1)).tolist() * nspec
    ydata = numpy.arange(float(nbins)).tolist() * nspec
    import time
    start = time.time()
    ws = CreateWorkspace(DataX = xdata, DataY = ydata, NSpec = nspec)
    print "%.1f MiB in %.3f sec" % (ws.getMemorySize() / 1000000.0, time.time() - start)
    return ws

def time_WorkspaceFactory(nbins, nspec):
    import numpy
    xdata = numpy.arange(float(nbins+1))
    ydata = numpy.arange(float(nbins))
    import time
    start = time.time()
    ws = WorkspaceFactory.create("Workspace2D", NVectors=nspec, XLength=nbins+1, YLength=nbins)
    for i in xrange(nspec):
        ws.setX(i, xdata)
        ws.setY(i, ydata)
    print "%.1f MiB in %.3f sec" % (ws.getMemorySize() / 1000000.0, time.time() - start)
    return ws
```
These functions create virtually identical workspaces. Data generation is not timed, so one would expect these functions to print comparable runtimes. Since `time_WorkspaceFactory` involves a loop, one might expect it to be slightly slower.

Here are some results from running them on my machine:
```
ws = time_WorkspaceFactory(1000, 1000)
24.0 MiB in 0.027 sec
ws = time_WorkspaceFactory(100, 10000)
24.0 MiB in 0.077 sec
ws = time_WorkspaceFactory(10000, 100)
24.0 MiB in 0.017 sec
ws = time_WorkspaceFactory(10000, 1000)
240.0 MiB in 0.157 sec
ws = time_WorkspaceFactory(100000, 1000)
2400.0 MiB in 1.428 sec

ws = time_CreateWorkspace(1000, 1000)
24.0 MiB in 5.021 sec
ws = time_CreateWorkspace(100, 10000)
24.0 MiB in 4.448 sec
ws = time_CreateWorkspace(10000, 100)
24.0 MiB in 5.415 sec
ws = time_CreateWorkspace(10000, 1000)
240.0 MiB in 58.488 sec
```
The `WorkspaceFactory` approach has reasonable runtimes that scale linearly with data size. Lower NSpec to NBins ratio, with the same data size, translates into less loop cycles and runs a bit faster.

The `CreateWorkspace` approach also scales linearly, but is incredibly slow in comparison. It's not clear why the high NSpec / low NBins version runs a bit faster compared to vice versa, but it is consistent about that.

After some testing, I found that commenting out the creation and usage of `AlgorithmHistory` in `Algorithm::logAlgorithmInfo` [here](https://github.com/mantidproject/mantid/blob/b2a841ed59d605271da83124e77698eaf90c8533/Framework/API/src/Algorithm.cpp#L1066-L1072) roughly doubled performance:
```
ws = time_CreateWorkspace(10000, 1000)
240.0 MiB in 27.555 sec
```

Additionally commenting out the `fillAlgorithmHistory` call in `Algorithm::execute` [here](https://github.com/mantidproject/mantid/blob/b2a841ed59d605271da83124e77698eaf90c8533/Framework/API/src/Algorithm.cpp#L565-L566) further boosted performance:
```
ws = time_CreateWorkspace(10000, 1000)
240.0 MiB in 1.483 sec
```
This is a huge improvement, but still about 10 times slower than the `WorkspaceFactory` approach.

Just commenting out those lines is not really an option, of course. A proper solution might involve treating large arrays as a special case in the AlgorithmHistory.

I haven't investigated where the remaining overhead is, but it seems to be before we get to `Algorithm::logAlgorithmInfo`.

OwenArnold, 2017-09-12:
@MikeHart85 have you had a look at what the workspace history looks like when generated? I suspect that this is IO bound, but would be good to know what it is attempting to write out as a point of reference.

MikeHart85, 2017-09-15:
In a 10x10 scenario, it looks like this:

```
CreateWorkspace(OutputWorkspace='ws', DataX='0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10,0,1,2,3,4,5,6,7,8,9,10', DataY='0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9', NSpec=10)
```

This is still quite managable at roughly 500 characters. 

But at 100x100, that becomes around 55,000 characters. gEdit isn't very happy when you try to paste that (I assume because it's a single line, because 55k isn't that much), but it manages.

At 1000x1000, this generates a 7.8 MB line and gEdit gives up completely. Even nano caps out a core for a few seconds.

The workspace history itself is fine, once the algorithm finishes running, because it only displays something like this: `0,1,2,3,4,5,6,7,8 ... ,997,998,999,1000`. On that note, that's also all you get in the clipboard if you Right-click -> Copy to Clipboard in the table (`Script to Clipboard` button gives the whole thing).

If you try to create the 1000x1000 one while the log is set to `debug`, Mantid freezes completely. `information` is fine. Presumably because it on `debug` it prints out every value without abbreviating it like above.

nvaytet, 2018-06-13:
@MikeHart85 If I add an arbitrary upper limit on the size of the property that is being written to the history
```cpp
    if (property->size() < 100) {
      m_properties.push_back(
          boost::make_shared<PropertyHistory>(property->createHistory()));
    }
```
in `AlgorithmHistory` [here](https://github.com/mantidproject/mantid/blob/b2a841ed59d605271da83124e77698eaf90c8533/Framework/API/src/AlgorithmHistory.cpp#L71-L72), then I get much better performance, without having to comment out the lines you suggested above (though still not as good as WorkspaceFactory).

```
ws = time_WorkspaceFactory(1000, 1000)
24.0 MiB in 0.022 sec
ws = time_WorkspaceFactory(100, 10000)
24.0 MiB in 0.082 sec
ws = time_WorkspaceFactory(10000, 100)
24.0 MiB in 0.009 sec
ws = time_WorkspaceFactory(10000, 1000)
240.0 MiB in 0.165 sec
ws = time_WorkspaceFactory(100000, 1000)
2400.0 MiB in 1.412 sec

ws = time_CreateWorkspace(1000, 1000)
24.0 MiB in 0.110 sec
ws = time_CreateWorkspace(100, 10000)
24.0 MiB in 0.127 sec
ws = time_CreateWorkspace(10000, 100)
24.0 MiB in 0.091 sec
ws = time_CreateWorkspace(10000, 1000)
240.0 MiB in 0.877 sec
```

This is of course not a viable solution because one needs to be able to re-create the workspace from running the history script. One would need a method to somehow compress the long strings or arrays that are in the DataX and DataY fields, in a way that the data can be re-constructed from them...

nvaytet, 2018-06-14:
I wonder if the slow-down comes from the fact that it is written to the history as a string? Couldn't we write it as an array of doubles and only convert it into a string if the history is requested by the user? That part would still be slow but at least the workspace creation would be fast. Then we just have to hope that the history is not requested all the time.

nvaytet, 2018-06-19:
So I think I finally tracked down where the bottleneck exactly is.

it's indeed during a conversion from the arrays of doubles containing DataX and DataY to strings of comma separated values. This occurs in the `join()` function inside the `Strings.h` file. It is using an `ostringstream` operator to write the array elements one by one to the return string, inserting a separator between each element. Optimizing the conversion to string might improve things but I am not sure we can get to something really fast if we want to keep the entire array in the history.

OwenArnold, 2018-06-19:
There a quite a few options for converting doubles to strings, so I would take a look at stack overflow to see what others have used.

nvaytet, 2018-06-20:
with doubles, the length of the string actually depends on what precision you want to use to output the results. If they are all round numbers, it's easy, but in principle this array of doubles can hold any numbers. And writing all of them to 14 decimal places is not workable I think... I guess I could try.

OwenArnold, 2018-06-20:
@nvaytet I think it would be very reasonable to set the precision. For example, saving loading parameter maps also involves fixing the precision. I'll see if I can get the details of that and post it here.

OwenArnold, 2018-06-20:
I couldn't find exactly what I was looking for but you will see that `std::setprecision` is quite common in Saving routines if you grep the codebase.

closed 2018-10-24 by nvaytet