mantidproject/mantid #20444: setX, setY, etc don't coerce data correctly from Python

MikeHart85, 2017-09-12:
### Summary

When calling `Workspace.setX` (etc) from Python, the data you pass in **must** be a numpy array of floats.

If it is not a numpy array, an exception is thrown. 
If it is a numpy array of non-floats, the resulting workspace has incorrect data.

### Details

Consider this test code:
```python
def generateWorkspace(nbins, nspec):
    import numpy
    xdata = numpy.arange(float(nbins+1))
    ydata = numpy.arange(float(nbins))
    ws = WorkspaceFactory.create("Workspace2D", NVectors=nspec, XLength=nbins+1, YLength=nbins)
    for i in xrange(nspec):
        ws.setX(i, xdata)
        ws.setY(i, ydata)
    return ws
```

This works as expected:
```python
ws = generateWorkspace(5, 5)

ws.extractX()
array([[ 0.,  1.,  2.,  3.,  4.,  5.],
       [ 0.,  1.,  2.,  3.,  4.,  5.],
       [ 0.,  1.,  2.,  3.,  4.,  5.],
       [ 0.,  1.,  2.,  3.,  4.,  5.],
       [ 0.,  1.,  2.,  3.,  4.,  5.]])

ws.extractY()
array([[ 0.,  1.,  2.,  3.,  4.],
       [ 0.,  1.,  2.,  3.,  4.],
       [ 0.,  1.,  2.,  3.,  4.],
       [ 0.,  1.,  2.,  3.,  4.],
       [ 0.,  1.,  2.,  3.,  4.]])
```

If we use lists instead of numpy arrays:
```python 
def generateWorkspace(nbins, nspec):
    xdata = range(nbins+1)
    ydata = range(nbins)
    ws = WorkspaceFactory.create("Workspace2D", NVectors=nspec, XLength=nbins+1, YLength=nbins)
    for i in xrange(nspec):
        ws.setX(i, xdata)
        ws.setY(i, ydata)
    return ws

ws = generateWorkspace(5, 5)
ArgumentError: Python argument types in
    MatrixWorkspace.setX(Workspace2D, int, list)
did not match C++ signature:
    setX(Mantid::API::MatrixWorkspace {lvalue} self, unsigned long workspaceIndex, Mantid::PythonInterface::NumPy::NdArray x)
```

If we don't cast to float:
```python
def generateWorkspace(nbins, nspec):
    import numpy
    xdata = numpy.arange(nbins+1)
    ydata = numpy.arange(nbins)
    ws = WorkspaceFactory.create("Workspace2D", NVectors=nspec, XLength=nbins+1, YLength=nbins)
    for i in xrange(nspec):
        ws.setX(i, xdata)
        ws.setY(i, ydata)
    return ws

ws = generateWorkspace(5, 5)

ws.extractX()
array([[  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323,   2.47032823e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323,   2.47032823e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323,   2.47032823e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323,   2.47032823e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323,   2.47032823e-323]])

ws.extractY()
array([[  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323],
       [  0.00000000e+000,   4.94065646e-324,   9.88131292e-324,
          1.48219694e-323,   1.97626258e-323]])

```

The last one is particularly bad, since it appears to work but silently corrupts the data. By the time you notice something is wrong, it may be difficult to trace it back to this as the cause.

OwenArnold, 2017-09-12:
@martyngigg any ideas on this?

martyngigg, 2017-09-12:
I think we need to be slightly more flexible [here](https://github.com/mantidproject/mantid/blob/253cb5f7e6b84d74a083c9e9295ce79f313db75b/Framework/PythonInterface/mantid/api/src/Exports/MatrixWorkspace.cpp#L72).

The function signature should needs to be relaxed from `Numpy::NdArray` to `object` and then the functions checks internally what to do with the argument whether it is a list or numpy array.

martyngigg, 2017-09-12:
I think we can improve that method to check for the type within the numpy array and perform the necessary coercion.

PR #21407 (2017-12-19): Coerce data correctly from Python Matrixworkspace::setXYE methods
**Description of work**

Fixed a bug in the data coercion logic when entering data through the python MatrixWorkspace::setX/setY/setE methods.

Also now allows lists as sources of the data.

**To test:**

See instructions in the issue for testing. The errors should now be gone and the data should be correct.

Fixes #20444

**Release Notes** 

Add note to [framework section](/mantidproject/mantid/commit/ddfd1969b569c6c39f7a96921b5a2537d4b69c09)

---

#### Reviewer ####

Please comment on the following ([full description](http://w [... cut]
closed 2017-12-19 by OwenArnold