mantidproject/mantid #20445: Algorithm timer shows incorrect runtime

MikeHart85, 2017-09-12:
When running an algorithm, the log outputs notices that look like this:
```
AlgorithmName started
AlgorithmName successful, Duration 0.01 seconds
```
The duration here can be extremely misleading.  For example:
```python
import time
start = time.time()
ws = CreateWorkspace(DataX=range(1000000), DataY=range(1000000), NSpec=1000)
print "%.2f sec" % (time.time() - start)
```
Output:
```
6.63 sec
```
Log:
```
CreateWorkspace started
CreateWorkspace successful, Duration 0.07 seconds
```
This happens because the timer is only measuring the `exec` of the algorithm, but there is considerable overhead (separate issue, see #20443) that occurs prior to that. Additionally, the "Algorithm started" message appears long before `exec` is called, so you will actually see the "started" notice, a long delay, followed by a claim of a short duration.

Since the overhead of calling an Algorithm can be so much heavier than the execution of the Algorithm itself, the duration of `exec` alone probably isn't very useful information and we should begin measuring the runtime much sooner.

In any case, a user would expect the timer to start when the "Algorithm started" notice is issued at the latest.

OwenArnold, 2017-09-12:
I agree most users would expect to see the total time incorporated. However, if there is considerable time being spent in the `init`, that is measurable for python/mantidsimplapi, but probably not for other gui usages, where you effectively have a 2-step process.

OwenArnold, 2017-09-12:
I think that in most cases the init time should be considerably smaller than the exec time. Is this `WorkspaceHistory` related?

MikeHart85, 2017-09-19:
As discussed earlier, the WorkspaceHistory issue makes this easy to see, but it is always measuring runtime incorrectly. This is because the timer starts far too late in `Algorithm::execute()`, long after the "Algorithm started." log notice has been issued.

PR #21627 (2018-01-24): Include time spent in Algorithm::execute in reported time.
See issue description. Also add debug output regarding time within execute.

**To test:**

Code review.

Fixes #21626.
Fixes #20445.

**Release Notes** 

Updated.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class in isolation?
- [ ]  [... cut]
closed 2018-01-24 by peterfpeterson