mantidproject/mantid #20448: CompareWorkspaces shows incomplete mismatch information

MikeHart85, 2017-09-12:
`CompareWorkspaces` compares two workspaces and produces two outputs:
- `Result` contains a boolean indicating whether the workspaces matched
- `Messages` contains a `TableWorkspace` of messages detailing the nature of any mismatches that were found

The table has three columns: Message, Workspace 1, Workspace 2

The Workspace columns are for indicating which workspaces didn't match (in case group workspaces are being compared). The Message column is supposed to contain a verbal description of the mismatch.

Instead of a full description of the mismatch, the Message column instead only contains the first word of the intended sentence. Such "Axis" instead of "Axis 0 type mismatch", or "Number" instead of "Number of spectra mismatch".

To test:
```python
ws1 = CreateWorkspace(DataX=[1], DataY=[1], NSpec=1)
ws2 = CreateWorkspace(DataX=[1], DataY=[2], NSpec=1)
result, msgs = CompareWorkspaces(ws1, ws2)

print result
False

print msgs.column(0)
['Data']
```
The issue seems to occur after or as a result of how `TableRow`s are added in `CompareWorkspaces::recordMismatch`, [here](https://github.com/mantidproject/mantid/blob/410d2c7c3339dc6260b325bfb2a1e17cab56753c/Framework/Algorithms/src/CompareWorkspaces.cpp#L1154). At that point, the message is still complete.

SimonHeybrock, 2018-06-06:
Related? https://github.com/mantidproject/mantid/issues/20400

MikeHart85, 2018-06-06:
Yes, looks like the same issue. Nice catch!

I've just tried it, and it looks like this was fixed by #20401, so I will close this.

It's strange that #20400 is still open. I think Sam might have reopened it because of #20487, but that is merged now too. I'm guessing #20400 should be closed as well, but might be best to check with Sam.

closed 2018-06-06 by MikeHart85