mantidproject/mantid #20508: Large number of Flake8 complexity warnings

josephframsay, 2017-09-19:
There are currently 41 warnings for too much cyclomatic complexity in Python scripts in the build servers. These should be reduced as much as possible. The existings warnings can be viewed on the Flake8 warnings page for any PR

mganeva, 2017-09-26:
@josephframsay could you please update the line about TOFTOF GUI to
`scripts/Interface/reduction_gui/reduction/toftof (1)` #20635?

josephframsay, 2017-09-26:
@mganeva done - thanks for your help on this!

martyngigg, 2017-09-27:
@josephframsay Is this meant for `3.12`?

josephframsay, 2017-09-28:
Yup - changed!

ElliotAOram, 2017-10-09:
Flake8 warnings from the algorithms section can  not be resolved until proper unit tests have been written. Issues have been opened to this affect see:
#20861
#20764

josephframsay, 2017-10-09:
Flake8 warnings from `scripts/reduction/reducer.py` were not resolved as the file has no unit tests on it. An issue has been open at #20863 to address this.

Other than that and `LRReflectivityOutput`, all files have either been refactored sufficient to make the warnings go away, or issued with a `noqa` flag to suppress the warnings on the build server. This has only been done in the case of files which are to be deprecated soon and have insufficient tests, making refactoring needlessly risky.

Once all the relevant PRs have gone through the cap on the Flake8 build server should presumably be lowered to the number of outstanding issues , which I think is 3.

closed 2017-10-09 by josephframsay