mantidproject/mantid #20566: Unscripted Testing ISIS Indirect interfaces

NickDraper, 2017-09-21:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See
* http://www.mantidproject.org/Indirect:ConvertToEnergy
* http://www.mantidproject.org/Indirect:Indirect_Data_Analysis
* http://www.mantidproject.org/UsingElwin
* http://www.mantidproject.org/UsingFury

josephframsay, 2017-09-22:
The files used for the tutorial in http://www.mantidproject.org/Indirect:ConvertToEnergy are not is the ISIS Sample Data folder

NickDraper, 2017-09-22:
Can you add the files to the zip file and give me or Martyn a copy to re upload.

Also feel free to improve the tutorial

josephframsay, 2017-09-22:
Yup, just making notes for myself!

josephframsay, 2017-09-26:
Tested on Windows and Ubuntu - other than the identified issue, no problems

closed 2017-09-26 by josephframsay