mantidproject/mantid #20569: Unscripted Testing Documentation

NickDraper, 2017-09-21:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Qt Assistant offline documentation
* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

AnthonyLim23, 2017-09-25:
Bivariate Normal does not render maths correctly 

Direct Inelastic Changes in release notes is missing maths

NickDraper, 2017-09-25:
Could you have a go at fixing these yourself?

Regards,
Nick Draper

AnthonyLim23, 2017-09-25:
I will. Just waiting to hear back from Anders to check if the T0 should be maths and if it should be T_0

AnthonyLim23, 2017-09-25:
In a PR now: #20631

closed 2017-09-25 by AnthonyLim23