mantidproject/mantid #20570: Unscripted Testing DGSReduction

NickDraper, 2017-09-21:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See http://www.mantidproject.org/Direct:DgsReduction
     and http://www.mantidproject.org/Create_MD_Workspace_GUI#Direct_Inelastic_Mod_Q

AndreiSavici, 2017-10-02:
The GUI is not working anymore 
```
ImportError: No module named Interface.reduction_gui.reduction.inelastic
  at line 10 in '<Interface>'
  caused by line 153 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_application.py'
  caused by line 343 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_application.py'
  caused by line 171 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_application.py'
  caused by line 40 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_gui/instruments/instrument_factory.py'
  caused by line 28 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_gui/instruments/dgs_interface_dev.py'
  caused by line 47 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_gui/widgets/inelastic/dgs_sample_setup.py'
  caused by line 38 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_gui/reduction/inelastic/dgs_sample_data_setup_script.py'
  caused by line 42 in '/home/3y9/Mantid/mantid/scripts/Interface/reduction_gui/reduction/inelastic/dgs_sample_data_setup_script.py'
```

The algorithm is being tested often, since on some instruments the autoreduction is using nightly

AndreiSavici, 2017-10-02:
With the changes in PR#20750, everything works as well as last time

PR #20750 (2017-10-02): Fixed the imports in DGS GUI
When starting the DgsReduction GUI in Interfaces->Direct, it should show up without any import errors. See issue #20570 for the output from nightly

Fixes #20570

**Release Notes** 

*Does not need to be in the release notes - bug was introduced when going to Python3*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standa [... cut]
closed 2017-10-03 by NickDraper