mantidproject/mantid #20575: Unscripted Testing Muon

NickDraper, 2017-09-21:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

As a minimum test http://www.mantidproject.org/Unscripted_Manual_Testing_MuonAnalysis.
For more info see http://www.mantidproject.org/Muon

AnthonyLim23, 2017-09-22:
there is a bug with ties #20552 and a crash issue #20546. These are both on my to do list next week.

AntonPiccardoSelg, 2017-09-25:
I have been testing script 2 and it has been segfaulting a lot on Ubuntu. Since test script 3 depends on the results of test script3, testing has become a bit difficult. Hence it would be good to modify the test instructions in a way that the tests are small and self-contained. 

The main issue I see is the spurious segfaults which I am getting. This seems to be a known issue.

closed 2017-09-25 by AntonPiccardoSelg

AnthonyLim23, 2017-09-25:
I should add, these problems are significantly more common on ubuntu than windows