mantidproject/mantid #20592: Unscripted Testing ISIS Reflectometry (old and new)

NickDraper, 2017-09-21:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

For testing procedures see http://www.mantidproject.org/Testing_ISIS_Reflectometry_GUI

gemmaguest, 2017-09-21:
Note that the testing documentation only refers to the old GUI. Please try to follow the same process for the new GUI, using the [new GUI documentation](http://docs.mantidproject.org/nightly/interfaces/ISIS%20Reflectometry.html) to help. Let me know if anything is unclear - it would be really useful for us to know of any stumbling blocks so that we can see what problems users might have when they transition to the new GUI.

samueljackson92, 2017-09-26:
Some minor points picked up from testing.
 - The `INTER_NR_test2.tbl` file from the documentation cannot be loaded into the interface. I does however appear to get loaded as a table workspace with the expected number of columns/rows. Perhaps it is just a bad file?

 - Setting uniform slicing (even or in seconds) to ridiculous values (e.g. `-1` or `qwerty`). does not cause an error message or anything to be displayed.

##
Niggly stuff from testing that might be worth thinking about in future (if someone ever gets round to it):

 - Font on the "Event Handling" tab on my screen is very small:

[image]

 - What's this? tool needs implementing for "Output Notebook" and "Process"

Will do some more testing and append stuff here as I find it.

gemmaguest, 2017-09-27:
I've created #20675 for the `.tbl` issues because I think these are regressions since Mantid 3.10

samueljackson92, 2017-10-03:
Apart from  #20675 I couldn't find anything else that was major, so I'm marking this one as closed.

closed 2017-10-03 by samueljackson92