mantidproject/mantid #20593: Unscripted Testing ISIS SANS (v2 GUI)

NickDraper, 2017-09-21:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Speak to Anton

gemmaguest, 2017-09-22:
Testing on Ubuntu 16.04

gemmaguest, 2017-09-28:
Here's the testing I've done so far:

- Open the interface and enter run number `74044` in the `SampleScatter` column. Press `Process`.
  - This gives `RuntimeError: There seems to have been an issue with setting the states. Make sure that a user filehas been loaded`. I think this is as expected.
- Load the `MaskFile.txt` user file from the sample data.
  - This doesn't give the same warnings that you usually get when loading this file in the original SANS interface.
  - The user file is not remembered when you re-open the interface, so you have to re-select it.
- Try processing again.
  - Gives `Trying to set _PositiveIntegerParameter#incident_monitor with an invalid value of -1` because a monitor has not been set.
  - The monitor seems to get set automatically in the old interface (or at least remembered from a previous run; again, this is not remembered if you close and re-open the new interface).
- Click `Settings` and `Adjustment` tab. Enter `2` for the `Incident monitor` and try processing again.
  - Gives `Property "INPUTWORKSPACE" is not set to a valid value: "Workspace __sans_dummy_gui_workspace is not of the correct type".`

gemmaguest, 2017-10-03:
Continuing testing following the above fixes (now testing on Windows 10), I got an error when the default instrument was different to that on the interface (I had OFFSPEC as the default and `LOQ` on the interface):
- With the same set up as the previous comment (using run 74044), check that the `Instrument` drop-down box in the interface says `LOQ`, and hit `Process` 
- You get the error `RuntimeError: Unable to find file: search object 74044`
- This works ok if the default instrument in MantidPlot is also `LOQ`

Another error I saw was this:
- I set the default instrument to `LOQ`, removed `loqdemo` from the search path and closed and re-opened the interface.
- Repeating the test above, I now got this confusing error (rather than an error about run 74044 not being found, which is what I'd expect): 
```
Error in execution of algorithm SANSGuiDataProcessorAlgorithm
Invalid value for property Filename (string) "DIRECT.041": File "DIRECT.041" not found
```

gemmaguest, 2017-10-03:
Apart from the issues noted above, the new interface seems to work well and produces the same results as the old interface.

AntonPiccardoSelg, 2017-10-03:
@gemmaguest Thanks for testing!

Regarding the first issue: This should have been fixed here: https://github.com/mantidproject/mantid/pull/20755

Regarding the second issue with `DIRECT.041`: You should see this issue since the direct file cannot be found. I don't quite know where the part `Invalid value for property Filename (string)` comes from, but otherwise I would expect that it complains about not finding the direct file.

closed 2017-10-04 by gemmaguest