mantidproject/mantid #20599: Procedure for error calculations used by Rebin2D is incorrect.

abuts, 2017-09-22:
### Expected behavior
The code:
```Python

Load(Filename=r'MAR22152_Ei15.00meV.nxspe', OutputWorkspace='Yb2_Ei15.00meV')

# # To generate 1D files for Yb2 data after background subtraction
SofQWNormalisedPolygon(InputWorkspace='Yb2_Ei15.00meV', OutputWorkspace='Yb2_Ei15.00meV_subt_SQW_coarse', QAxisBinning='0,0.001,5', EMode='Direct')
#SofQWCentre(InputWorkspace='Yb2_Ei15.00meV_subt', OutputWorkspace='Yb2_Ei15.00meV_subt_SQW_coarse', QAxisBinning='0,0.001,5', EMode='Direct')
SofQWNormalisedPolygon(InputWorkspace='Yb2_Ei15.00meV', OutputWorkspace='Yb2_Ei15.00meV_subt_SQW_fine', QAxisBinning='0,0.01,5', EMode='Direct')
#SofQWCentre(InputWorkspace='Yb2_Ei15.00meV_subt', OutputWorkspace='Yb2_Ei15.00meV_subt_SQW_coarse', QAxisBinning='0,0.001,5', EMode='Direct')

Rebin2D(InputWorkspace='Yb2_Ei15.00meV_subt_SQW_coarse', OutputWorkspace='Yb2_Ei15.00meV_bcksubt_1D_coarseQ', Axis1Binning='-7,0.02,15', Axis2Binning='0,3,3', UseFractionalArea=True)
Rebin2D(InputWorkspace='Yb2_Ei15.00meV_subt_SQW_fine', OutputWorkspace='Yb2_Ei15.00meV_bcksubt_1D_fineQ', Axis1Binning='-7,0.02,15', Axis2Binning='0,3,3', UseFractionalArea=True)

```
calculates signal and error along the trajectory, chosen by user with different accuracy. 
The data file used by the script can be downloaded from:

One would expect the signal and error calculated by the script to be independent on the resolution

### Actual behaviour

While the signal does not depend on the resolution the error do. In fact, changing resolution one can get almost any value for error.  In addition to that, the calculated error is substantially different from the one calculated by reference code (Mslice)

### Steps to reproduce the behaviour
See above
### Platforms affected
any.

AndreiSavici, 2017-09-27:
This is not a bug. SofQW... algorithms calculate an intensity that is normalized. Different bins have different statistical weights. Once the data is normalized, Rebin2D will average the data as if it were the same statistical weights. In this case the output signal and error are going to be dominated by regions with poor statistics.

Mslice goes back to the NXSPE data when you rebin, so if all contributions in the NXSPE file(s) are equally weighted, the statistics of the output is OK.

In conclusion: NEVER use Rebin2D on data generated by SofQW...  algorithms.

I suggest this issue be closed

abuts, 2017-09-28:
I like the simplicity of this approach and instinctively was trying to follow it when this issue was raised with me, but suspect that Tatiana ( tatiana.guidi@stfc.ac.uk) and other users would be not satisfied. 

Pair SofQW and Rebin2D are used in sliceviewer for analysing powder images. I suspect @mducle uses it in Mantid Mslice and this one is intended for very inexperience users, who will be surprised to see different errors for the same data. 

I am happy if it fine with Tatiana and @mducle, but afraid they will disagree and either SofQW or Rebin2D should be modified appropriately.

mducle, 2017-09-28:
@abuts @AndreiSavici Python MSlice (the standalone GUI) doesn't use `Rebin2D`. I think this is only used in the `SliceViewer`. 

>  In conclusion: NEVER use Rebin2D on data generated by SofQW... algorithms.

As @abuts says, I don't agree with this - the `SliceViewer` is used extensively, and for powder INS data, we always need to use `SofQW` - so either we need to change `SliceViewer` or `SofQW` or `Rebin2D` - maybe the best option is either to create a new `Rebin2DWeighted` for use in `SliceViewer`, or to add an option to take into account the statistical weights (I guess we can take it to be inversely proportional to the variance which is stored in the `MatrixWorkspace`.) in `Rebin2D`.

Now, I guess we will eventually move everyone over to the Python MSlice clone (which doesn't use `Rebin2D` and only uses `SofQW3` for data which has been rebinned using a rings mapping of detector pixels or for MARI which doesn't have PSDs. In this case `SofQW3` is called once with a fixed Q-bin size [the Q-bin size is inferred from the 2theta spacing data if an instrument name is defined, otherwise the Q-range is divided into 100 bins]), but `SliceViewer` is still widely used now, so I think we still need to address this issue.

AndreiSavici, 2017-09-28:
@abuts @mducle Does not matter if you like it or not, normalized data should not be rebined. DAVE Mslice works the same way as Mantid with MD workspaces. It carries separately the data and normalization. If you use ConvertToMD on a workspace2D (NXSPE), each histogram bin is assumed to be equally significant. If that is the case, the total signal in the MDBox needs to be divided by the number of contributions, in order to get S(|Q|,w) - that's why we use number of event normalization. So if you want to change something you have several options:
1. Use MDEvents from NXSPE 
2. Modify SofQW... to output two workspaces (data and normalization) that you need to divide when you want to visualize/use them. Then Rebin2D would need to be applied to both of them, and divide the two outputs. SliceViewer will also need to be made aware of this changes. This is definitely not something that will get into 3.11

abuts, 2017-10-03:
After investigating the problem and discussing it with @mducle, I suspect that the issue is much simple -- the errors, produced by SofQW normalized polygons are the same regardless of the Q-binning selected. This is certainly a bug.

mducle, 2017-10-04:
Ok. I think I've found the problem - it's actually a case of using the *errors* (standard deviations) in a calculation rather than the *variance*. 

So, actually, the conceptual issue raised by @AndreiSavici doesn't apply here - in fact option 2 in the above comment is already implemented in Mantid: There is a derived class `RebinnedOutput` from `Workspace2D` which carries around within it the fractional areas of bins / normalisation constants. `SofQWNormalisedPolygons` always outputs a `RebinnedOutput` workspace with the fractional area information. 

If you use `Rebin2D` with the `UseFractionalArea` flag set to `True` then it will take into account the bin fractional area (normalisation) information - and return the correct intensity for a cut/slice (but not the correct error due to scaling the error rather than the variance by the fractional area). If `UseFractionalArea=False` is set, then `Rebin2D` returns a plain `Workspace2D` and discards the normalisation information and it will give both *wrong intensities and wrong errors*. 

Now, *by default **UseFractionalArea=False*** because `Rebin2D` also operates on `Workspace2D` (which I guess is assumed to be the default) which has no normalisation information. Since using the fractional area information is the correct way to handle a `RebinnedOutput` workspace - could we change the behaviour to **always** use this information if it is available? E.g. I'm proposing to **remove `UseFractionalArea` as an option and act as if `UseFractionalArea=True` for `RebinnedOutput` workspaces and `UseFractionalArea=False` for `Workspace2D`**. This basically assumes that users are stupid and will chose the wrong option (`UseFractionalArea=False` for the output of `SofQWNormalisedPolygons` because it's the default). But, to be honest, I didn't really understood this distinction until I looked at the code today, even though I usually set `UseFractionalArea=True` because it sounded more accurate...

Anyway, the actual bug is in the function ` FractionalRebinning::rebinToFractionalOutput()` in [FractionalRebinning.cpp line 218](https://github.com/mantidproject/mantid/blob/master/Framework/DataObjects/src/FractionalRebinning.cpp#L218), where the *errors* (standard deviations) are weighted by the fractional overlap area rather than the *variance*:

```C++
if (intersection(outputQ, inputQ, intersectOverlap)) {
  const double weight = intersectOverlap.area() / inputQ.area();
  yValue *= weight;
  double eValue = inE[j] * weight;
  ...
  eValue *= eValue;
  PARALLEL_CRITICAL(overlap) {
    outputWS->mutableY(yi)[xi] += yValue;
    outputWS->mutableE(yi)[xi] += eValue;
    outputWS->dataF(yi)[xi] += weight;
  }
}
```

Thus the `weight` appears as `weight*weight` in the output. For finer binning, the weights for each bin will be smaller so when squared this will give a smaller apparent errorbar. The problem is that just changing line 218 to `double eValue = (inE[j] * inE[j]) * weight;` and removing line 227 (`eValue *= eValue;`) doesn't work, because at the end of `SofQWNormalisedPolygons()` the code scales the *variance* by the total weight (in `RebinnedOutput::finalize()` and *then* takes the square-root (in `FractionalRebinning::normaliseOutput()`). This makes the output of `SofQWNormalisedPolygons` look ok in a graph, but mangles the errors for further computation. (There is even a note in [RebinnedOutput.cpp line 94](https://github.com/mantidproject/mantid/blob/master/Framework/DataObjects/src/RebinnedOutput.cpp#L94) that this unmangling should happen before further rebin or integration!).

So, I can see a couple of ways to unmangle this:

1. Unmangle in `Rebin2D` - this means creating a copy of the input workspace, changing the errors array of the copy to the unmangled version and then carrying on the calculation as before. This is probably the most straightforward to implement, but results in a memory penalty.
2. Unmangle in `FractionalRebinning` - this can be done in-place, but will need more extensive restructuring of the code (overloading `rebinToFractionalOutput()` to take both a `RebinnedOutput` workspace as the input workspace as well as a normal `MatrixWorkspace`/`Workspace2D`). 
3. Adding an `unfinalize()` method to `RebinnedOutput` which would undo the scaling by the fractional areas in a workspace. `Rebin2D` could then call `unfinalize()` on the *input*, carry out its calculations and the call `finalize()` again on the *input* at the end. This is probably the worst option, since it essentially doubles the amount of computation, and messes with the *input* workspace; but it requires minimal changes to `Rebin2D` and `FractionalRebinning` and has no memory penalty.

I'm inclined to go with option 2 - what do you @martyngigg @abuts think?

AndreiSavici, 2017-10-04:
@mducle Nice. I was not aware that we keep the weights.
If you want this to be in 3.11, I think the one with the least amount of problems is option 1. I would then implement option 2 in the next release.

mducle, 2017-10-05:
So it turned out not to be too hard to implement option 2. I just needed to use a cast from `MatrixWorkspace` to `RebinnedOutput`. I'm running unit and system tests now. So far there's a problem with the `SofQWNormalisedPolygonTest` and `ComputeIncoherentDOSTest`...

PR #21925 (2018-03-08): Missing fractional propagation code in `FractionalRebinning`
During the merge of master into the branch for #20953 the code which propagates the fractional areas in a re-rebin was lost. This PR adds it again. Without this code, the actual example code in #20599 would still give the incorrect errors.

**To test:**

Compile and run the test code in #20599. Check the signal and errors in the two `*_bcksubt_1D_*` workspaces are the same (within a few percent).

Fixes #20599

**Release Notes** 

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([ [... cut]
closed 2018-02-27 by peterfpeterson