mantidproject/mantid #20610: Cannot save or plot result in Indirect Elwin interface

josephframsay, 2017-09-22:
### Expected behavior
After loading and running a nexus file in the Elwin tab of the Indirect Data Analysis GUI, Plot Result would be expected to plot the output data, and Save Result would be expected to save it to an external file.

### Actual behavior
Neither operation is successful - in both cases we get a 'workspace not found' error.

### Steps to reproduce the behavior
Go to `Interfaces->Indirect->Indirect Data Analysis` and, inside the Elwin tab, load a focused IRIS file (such as [this](https://github.com/mantidproject/mantid/files/1324565/irs26176_graphite002_red.zip). Click Run, and then try to do Plot Result and Save Result.

### Platforms affected

martyngigg, 2017-09-25:
So it looks to me here that the plots get created and the files get saved but as there is no `_elt` workspace then it warns that this specific one is not there and then the plots get pushed back behind the main window.

I can't seem to get it to create the `_elt` workspace, I assume it is optional? - @SpencerHowells, what do I need to do? Also, if this is optional then we should probably have a less aggressive warning for the user. Does this seem appropriate?

SpencerHowells, 2017-09-25:
@martyngigg The _elt ws is created when the option to normalise to lowest temperature is selected. So it should just have a check that it exists with no warning.

martyngigg, 2017-09-26:
@SpencerHowells Thanks. I had checked the option but it still didn't produce an `_elt` workspace. It turned out I was using an old file without any logs in it.

PR #20644 (2017-09-27): Indirect Data Analysis: Suppress warning on missing optional output 
Description of work.

**To test:**

See the test instructions in the original issue. The missing `_elt` workspace should now not produce a warning. To produce the `_elt` workspace first use the Indirect Data Reduction interface to reduce a newer file such as `IRIS00073585` and save the result. Load this into the data analysis interface, click "Normalise to Lowest Temperature" and click run. The `_elt` workspace should now appear and plotting should show all 4 plots.

Fixes #20610 

**Release Notes** 

*Feature introduced this [... cut]
closed 2021-03-03 by stale[bot]