mantidproject/mantid #20662: Table locking columns via the scripting window

AnthonyLim23, 2017-09-27:
It has been reported from users that the results tables can have different columns set to editable/read only. This is a feature they would like to be exposed to the scripting window, so they could create a table with some read only columns.

AnthonyLim23, 2017-09-28:
a user can change if a column is "read only" by going to table->table options. However, this needs to be done every time the table is opened.

AnthonyLim23, 2021-03-01:
might be worth looking at. What do you think @martyngigg

martyngigg, 2021-03-01:
It looks like we have the access on the C++ side already so it would just be a matter of wiring it up. Probably a good induction issue.

AnthonyLim23, 2021-03-01:
Added the label so it doesnt get lost

martyngigg, 2021-09-06:
As this is an induction issue I'm going to mark this never stale as it's useful to have some of these in the pot.