mantidproject/mantid #20672: Umbrella: Python3 compatibility in doc tests

josephframsay, 2017-09-27:
As part of the move to Python3, many of the doc tests need to be refactored slightly in order to be compatible. It is expected that this will mostly be a matter of changing to the new functional-style `print` statements.

Please pick a chunk of 5 to 10 algorithm at a time:
1. Pick algorithms from the **top**.
2. Write a comment on this issue, listing the algorithms you picked. Also move those algorithms to the *Assigned/Done* list
3. Make sure no one picked the same algorithm while writing the comment (refresh page).
4. Create an issue, start refactoring, and create a PR. See below for a template for copy and paste to make creating the issue quicker.
5. Do not hesitate to create the PR early for an intermediate review

Issue template:

```
Python3 doctest compatibility: algorithms <X to Y>

This issue is part of the move to Python3 in the doctests, 
Algorithms covered in this issue are:
```

PR template (edit as needed):

```
Python3 doctest compatibility: algorithms <X to Y>

## Description of work.
This PR is part of refactoring effort to bring Python3 compatiblity into the doctests.
See  for details.

The following tests were refactored to be compatible :

## To test.

Code review.

Fixes <your issue>.
Internal change, no release notes
```

## Algorithms assigned/done

```
```

## Algorithms to do
```
    Abins
    AbortRemoteJob
    AbsorptionCorrection
    AccumulateMD
    AddLogDerivative
    AddNote
    AddPeak
    AddPeakHKL
    AddSampleLog
    AddSampleLogMultiple
    AddTimeSeriesLog
    AlignAndFocusPowder
    AlignAndFocusPowderFromFiles
    AlignComponents
    AlignDetectors
    AlphaCalc
    AndMD
    AngularAutoCorrelationsSingleAxis
    AngularAutoCorrelationsTwoAxes
    AnnularRingAbsorption
    AnvredCorrection
    AppendGeometryToSNSNexus
    AppendSpectra
    ApplyCalibration
    ApplyDeadTimeCorr
    ApplyDetailedBalance
[... 790 lines cut ...]
    UpdateInstrumentFromFile
    UpdatePeakParameterTableValue
    UpdateScriptRepository
    UploadRemoteFile
    UserFunction1D
    VelocityAutoCorrelations
    VelocityCrossCorrelations
    VesuvioCalculateGammaBackground
    VesuvioCalculateMS
    VesuvioCorrections
    VesuvioDiffractionReduction
    VesuvioL1ThetaResolution
    VesuvioPreFit
    VesuvioResolution
    VesuvioTOFFit
    VesuvioThickness
    ViewBOA
    VisionReduction
    WeightedMean
    WeightedMeanMD
    WeightedMeanOfWorkspace
    WienerSmooth
    WorkflowAlgorithmRunner
    XorMD
```

josephframsay, 2017-09-27:
```
    Abins
    AbortRemoteJob
    AbsorptionCorrection
    AccumulateMD
    AddLogDerivative
    AddNote
    AddPeak
    AddPeakHKL
```

samueljackson92, 2017-09-27:
#20674 20674

```
BASISReduction
    BASISReduction311
    BayesQuasi
    BayesStretch
    Bin2DPowderDiffraction
    BinMD
    BinWidthAtX
    BinaryOperateMasks
    CalMuonDeadTime
    CalMuonDetectorPhases
```

josephframsay, 2017-09-28:
```

    BASISReduction
    BASISReduction311
    BayesQuasi
    BayesStretch
    Bin2DPowderDiffraction
    BinMD
    BinWidthAtX
    BinaryOperateMasks
    CalMuonDeadTime
    CalMuonDetectorPhases
    CalculateChiSquared
    CalculateCostFunction
    CalculateCountRate
    CalculateCoverageDGS
    CalculateDIFC
    CalculateEfficiency
    CalculateFlatBackground
    CalculateMuonAsymmetry
    CalculatePeaksHKL
    CalculateSampleTransmission
    CalculateSlits
```

josephframsay, 2017-09-28:
ConjoinFiles
    ConjoinSpectra
    ConjoinWorkspaces
    ConjoinXRuns
    ConvertAxesToRealSpace
    ConvertAxisByFormula
    ConvertCWPDMDToSpectra
    ConvertCWSDExpToMomentum
    ConvertCWSDMDtoHKL
    ConvertDiffCal
    ConvertEmptyToTof
    ConvertFromDistribution
    ConvertMDHistoToMatrixWorkspace
    ConvertMultipleRunsToSingleCrystalMD
    ConvertSnsRoiFileToMask
    ConvertSpectrumAxis
    ConvertSpiceDataToRealSpace
    ConvertTableToMatrixWorkspace
    ConvertToConstantL2
    ConvertToDetectorFaceMD
    ConvertToDiffractionMDWorkspace
    ConvertToDistribution
    ConvertToEventWorkspace
    ConvertToHistogram
    ConvertToMD
    ConvertToMDMinMaxGlobal
    ConvertToMDMinMaxLocal
    ConvertToMatrixWorkspace
    ConvertToPointData
    ConvertToReflectometryQ
    ConvertToYSpace
    ConvertUnits
    ConvertUnitsUsingDetectorTable
    ConvolutionFitSequential
    ConvolveWorkspaces

josephframsay, 2017-09-29:
CopyDetectorMapping
    CopyInstrumentParameters
    CopyLogs
    CopySample
    CorelliCrossCorrelate
    CorrectKiKf
    CorrectLogTimes
    CorrectTOF
    CorrectTOFAxis
    CorrectToFile
    CountReflections
    CreateCacheFilename
    CreateCalFileByNames
    CreateChopperModel
    CreateChunkingFromInstrument
    CreateDummyCalFile
    CreateEPP
    CreateEmptyTableWorkspace
    CreateFlatEventWorkspace
    CreateGroupingWorkspace
    CreateLeBailFitInput
    CreateLogPropertyTable
    CreateLogTimeCorrection
    CreateMD
    CreateMDHistoWorkspace
    CreateMDWorkspace
    CreateModeratorModel
    CreatePSDBleedMask
    CreatePeaksWorkspace
    CreateSampleShape
    CreateSampleWorkspace
    CreateSimulationWorkspace
    CreateSingleValuedWorkspace
    CreateTransmissionWorkspace
    CreateTransmissionWorkspaceAuto
    CreateUserDefinedBackground
    CreateWorkspace

soininen, 2017-09-29:
```
    CropToComponent
    CropWorkspace
    CrossCorrelate
    CrystalFieldEnergies
    CuboidGaugeVolumeAbsorption
    CutMD
    CylinderAbsorption
    CylinderPaalmanPingsCorrection
    DNSComputeDetEffCorrCoefs
    DNSFlippingRatioCorr
    DNSMergeRuns
    DPDFreduction
```

edwardb96, 2017-09-29:
#20715 
```DakotaChiSquared
    DefineGaugeVolume
    DeleteLog
    DeleteTableRows
    DeleteWorkspace
    DeleteWorkspaces
    DeltaPDF3D
    DetectorDiagnostic
    DetectorEfficiencyCor
    DetectorEfficiencyCorUser
    DetectorEfficiencyVariation
    DetectorFloodWeighting
    DetermineChunking
    DgsAbsoluteUnitsReduction
    DgsConvertToEnergyTransfer
    DgsDiagnose
    DgsPreprocessData
    DgsProcessDetectorVanadium
    DgsReduction
    DgsRemap
    DiffPeaksWorkspaces
    DiffractionEventCalibrateDetectors
    DiffractionFocussing
    DirectILLApplySelfShielding
    DirectILLCollectData
    DirectILLDiagnostics
    DirectILLIntegrateVanadium
    DirectILLReduction
    DirectILLSelfShielding
    Divide
    DivideMD
    DownloadFile
    DownloadInstrument
    DownloadRemoteFile```

josephframsay, 2017-09-29:
EQSANSAzimuthalAverage1D
    EQSANSDarkCurrentSubtraction
    EQSANSDirectBeamTransmission
    EQSANSLoad
    EQSANSMonitorTOF
    EQSANSNormalise
    EQSANSPatchSensitivity
    EQSANSQ2D
    EQSANSResolution
    EQSANSTofStructure
    EditInstrumentGeometry
    ElasticWindow
    ElasticWindowMultiple
    EnergyWindowScan
    EnggCalibrate
    EnggCalibrateFull
    EnggFitDIFCFromPeaks
    EnggFitPeaks
    EnggFocus
    EnggVanadiumCorrections

josephframsay, 2017-09-29:
FFT
    FFTDerivative
    FFTSmooth
    FakeISISEventDAE
    FakeISISHistoDAE
    FakeMDEventData
    FilterBadPulses
    FilterByLogValue
    FilterByTime
    FilterByXValue
    FilterEvents
    FilterEventsByLogValuePreNexus
    FilterLogByTime
    FilterPeaks
    FindCenterOfMassPosition
    FindClusterFaces
    FindDeadDetectors
    FindDetectorsInShape
    FindDetectorsOutsideLimits
    FindDetectorsPar
    FindEPP
    FindPeakBackground
    FindPeaks
    FindPeaksMD
    FindReflectometryLines
    FindSXPeaks
    FindUBUsingFFT
    FindUBUsingIndexedPeaks
    FindUBUsingLatticeParameters
    FindUBUsingMinMaxD
    Fit
    FitGaussian
    FitPeak
    FitPowderDiffPeaks
    FitResolutionConvolvedModel
    FixGSASInstrumentFile
    FlatPlateAbsorption
    FlatPlatePaalmanPingsCorrection

samueljackson92, 2017-10-03:
#20759

```
    GSASIIRefineFitPeaks
    GeneralisedSecondDifference
    GenerateEventsFilter
    GenerateGroupingPowder
    GenerateGroupingSNSInelastic
    GenerateIPythonNotebook
    GeneratePeaks
    GeneratePythonScript
    GetAllEi
    GetDetOffsetsMultiPeaks
    GetDetectorOffsets
    GetEi
    GetEiMonDet
    GetEiT0atSNS
    GetIPTS
    GetQsInQENSData
    GetSpiceDataRawCountsFromMD
    GetTimeSeriesLogInformation
    GoniometerAnglesFromPhiRotation
    GreaterThanMD
    GroupDetectors
    GroupWorkspaces
```

ElliotAOram, 2017-10-03:
``` 
    HFIRDarkCurrentSubtraction
    HFIRLoad
    HFIRSANSNormalise
    HFIRSANSReduction
    HRPDSlabCanAbsorption
    HasUB
    He3TubeEfficiency
    IQTransform
    ISISIndirectDiffractionReduction
    ISISIndirectEnergyTransfer
    IdentifyNoisyDetectors
    ImportMDEventWorkspace
    ImportMDHistoWorkspace
    IndexPeaks
    IndexSXPeaks
    IndirectAnnulusAbsorption
    IndirectCalibration
    IndirectCylinderAbsorption
    IndirectDiffScan
    IndirectFlatPlateAbsorption
    IndirectILLEnergyTransfer
    IndirectILLReductionFWS
    IndirectILLReductionQENS
    IndirectQuickRun
    IndirectResolution
    IndirectSampleChanger
    IndirectTransmission
    IndirectTransmissionMonitor
```

soininen, 2017-10-04:
```
    IntegrateByComponent
    IntegrateEPP
    IntegrateEllipsoids
    IntegrateEllipsoidsTwoStep
    IntegrateFlux
    IntegrateMDHistoWorkspace
    IntegratePeakTimeSlices
    IntegratePeaksCWSD
    IntegratePeaksHybrid
    IntegratePeaksMD
    IntegratePeaksMDHKL
    IntegratePeaksUsingClusters
    Integration
```

josephframsay, 2017-10-04:
InterpolatingRebin
    InvertMDDim
    InvertMask
    IqtFitMultiple
    IqtFitSequential
    LRAutoReduction
    LRDirectBeamSort
    LRPeakSelection
    LRPrimaryFraction
    LRReflectivityOutput
    LRScalingFactors
    LRSubtractAverageBackground
    LeBailFit
    LessThanMD
    LineProfile
    LiquidsReflectometryReduction

josephframsay, 2017-10-04:
```
Load-v1
LoadAscii-v2
LoadBBY-v1
LoadCIF-v1
LoadCalFile-v1
LoadCanSAS1D-v1
LoadCanSAS1D-v2
LoadDNSLegacy-v1
LoadDetectorInfo-v1
LoadDetectorsGroupingFile-v1
LoadDspacemap-v1
LoadEmptyInstrument-v1
LoadEmptyVesuvio-v1
LoadEventAndCompress-v1
LoadEventNexus-v1
LoadFITS-v1
LoadFlexiNexus-v1
LoadFullprofFile-v1
LoadFullprofResolution-v1
LoadGSASInstrumentFile-v1
LoadHKL-v1
```

josephframsay, 2017-10-04:
LoadIDFFromNexus-v1
LoadILLDiffraction-v1
LoadILLIndirect-v2
LoadILLReflectometry-v1
```
LoadILLSANS-v1
LoadILLTOF-v2
LoadISISNexus-v2
LoadInstrument-v1
LoadInstrumentFromNexus-v1
LoadInstrumentFromRaw-v1
LoadIsawDetCal-v1
LoadIsawPeaks-v1
LoadIsawSpectrum-v1
LoadIsawUB-v1
LoadLLB-v1
LoadLog-v1
LoadLogPropertyTable-v1
```

josephframsay, 2017-10-04:
```
LoadMD-v1
LoadMLZ-v1
LoadMappingTable-v1
LoadMask-v1
LoadMcStas-v1
LoadMultipleGSS-v1
LoadMuonLog-v1
LoadMuonNexus-v2
LoadNMoldyn3Ascii-v1
LoadNMoldyn4Ascii-v1
LoadNMoldyn4Ascii1D-v1
LoadNXcanSAS-v1
LoadNexus-v1
LoadNexusLogs-v1
LoadNexusMonitors-v1
LoadNexusMonitors-v2
LoadNexusProcessed-v1
LoadParameterFile-v1
LoadPreNexusLive-v1
LoadPreNexusMonitors-v1
```

josephframsay, 2017-10-05:
```
LoadRKH-v1
LoadRaw-v3
LoadRawBin0-v1
LoadRawSpectrum0-v1
LoadSESANS-v1
LoadSINQ-v1
LoadSINQFile-v1
LoadSINQFocus-v1
LoadSPE-v1
LoadSQW-v1
LoadSampleDetailsFromRaw-v1
LoadSassena-v1
LoadSpiceAscii-v1
LoadSpiceXML2DDet-v1
LoadVTK-v1
LoadVesuvio-v1
LoadVisionElasticBS-v1
LoadVisionElasticEQ-v1
LoadVisionInelastic-v1
LoadVulcanCalFile-v1
```

josephframsay, 2017-10-05:
```

MaskWorkspaceToCalFile-v1
MatchPeaks-v1
Max-v1
MaxEnt-v1
MaxMin-v1
MayersSampleCorrection-v1
Mean-v1
MedianBinWidth-v1
MergeCalFiles-v1
MergeLogs-v1
MergeMD-v1
MergeRuns-v1
Min-v1
Minus-v1
```

josephframsay, 2017-10-05:
```

ModeratorTzero-v1
ModeratorTzeroLinear-v1
ModifyDetectorDotDatFile-v1
MolDyn-v1
MonitorEfficiencyCorUser-v1
MonteCarloAbsorption-v1
MoveInstrumentComponent-v1
MultipleScatteringCylinderAbsorption-v1
Multiply-v1
MultiplyRange-v1
MuonGroupDetectors-v1
MuonProcess-v1
MuscatData-v1
MuscatFunc-v1
MuscatSofQW-v1
```

josephframsay, 2017-10-05:
```
NMoldyn4Interpolation-v1
NRCalculateSlitResolution-v1
NormaliseByCurrent-v1
NormaliseByDetector-v1
NormaliseByPeakArea-v1
NormaliseByThickness-v1
NormaliseSpectra-v1
NormaliseToMonitor-v1
NormaliseToUnity-v1
NormaliseVanadium-v1
OSIRISDiffractionReduction-v1
OneMinusExponentialCor-v1
OptimizeCrystalPlacement-v1
OptimizeLatticeForCellType-v1
PDCalibration-v1
PDFFourierTransform-v1
PDLoadCharacterizations-v1
```

brandonhewer, 2017-10-05:
Issue #20802
```
Logarithm-v1
LorentzCorrection-v1
MDHistoToWorkspace2D-v1
MDNormDirectSC-v1
MDNormSCD-v1
MDNormSCDPreprocessIncoherent-v1
MSDFit-v1
MagFormFactorCorrection-v1
MaskAngle-v1
MaskBTP-v1
MaskBins-v1
MaskBinsFromTable-v1
MaskDetectors-v1
MaskDetectorsInShape-v1
MaskPeaksWorkspace-v1
```

brandonhewer, 2017-10-05:
Issue #20811 
```
PaddingAndApodization-v1
Pause-v1
PawleyFit-v1
PeakIntensityVsRadius-v1
PeaksInRegion-v1
PeaksOnSurface-v1
PerformIndexOperations-v1
PhaseQuad-v1
PlotAsymmetryByLogValue-v1
PlotPeakByLogValue-v1
Plus-v1
PoissonErrors-v1
PoldiAnalyseResiduals-v1
PoldiAutoCorrelation-v5
PoldiCreatePeaksFromCell-v1
```

brandonhewer, 2017-10-05:
Issue #20815 
```
PoldiCreatePeaksFromFile-v1
PoldiDataAnalysis-v1
PoldiFitPeaks1D-v1
PoldiFitPeaks1D-v2
PoldiFitPeaks2D-v1
PoldiIndexKnownCompounds-v1
PoldiLoadRuns-v1
PoldiMerge-v1
PoldiPeakSearch-v1
PoldiPeakSummary-v1
PoldiTruncateData-v1
PolynomialCorrection-v1
Power-v1
PowerLawCorrection-v1
PredictFractionalPeaks-v1
```

josephframsay, 2017-10-05:
```
PredictPeaks-v1
PreprocessDetectorsToMD-v1
ProcessBackground-v1
ProcessIndirectFitParameters-v1
ProjectMD-v1
QueryMDWorkspace-v1
RRFMuon-v1
RayTracerTester-v1
ReadGroupsFromFile-v1
RealFFT-v1
Rebin-v1
Rebin2D-v1
RebinByPulseTimes-v1
Rebunch-v1
RecordPythonScript-v1
ReflectometryReductionOne-v1
ReflectometryReductionOne-v2
ReflectometryReductionOneAuto-v1
ReflectometryReductionOneAuto-v2
Regroup-v1
```

brandonhewer, 2017-10-05:
Issue #20818 
```
RemoveBackground-v1
RemoveBins-v1
RemoveExpDecay-v1
RemoveLogs-v1
RemoveMaskedSpectra-v1
RemovePromptPulse-v1
RemoveWorkspaceHistory-v1
RenameLog-v1
RenameWorkspace-v1
RenameWorkspaces-v1
ReplaceSpecialValues-v1
ReplicateMD-v1
ResNorm-v1
ResNorm-v2
ResizeRectangularDetector-v1
```

brandonhewer, 2017-10-05:
Issue #20820
```
RetrieveRunInfo-v1
RingProfile-v1
RotateInstrumentComponent-v1
RotateSource-v1
RunPythonScript-v1
SANSBeamFluxCorrection-v1
SANSConvertToWavelengthAndRebin-v1
SANSDarkRunBackgroundCorrection-v1
SANSFitShiftScale-v1
SANSStitch-v1
SANSWideAngleCorrection-v1
SCDCalibratePanels-v1
SassenaFFT-v1
SaveANSTOAscii-v1
SaveAscii-v2
```

brandonhewer, 2017-10-05:
Issue #20823

```
SaveCSV-v1
SaveCanSAS1D-v1
SaveCanSAS1D-v2
SaveDaveGrp-v1
SaveDetectorsGrouping-v1
SaveDiffFittingAscii-v1
SaveDspacemap-v1
SaveFITS-v1
SaveFocusedXYE-v1
SaveFullprofResolution-v1
SaveGSASInstrumentFile-v1
SaveGSS-v1
SaveHKL-v1
SaveILLCosmosAscii-v1
SaveISISNexus-v1
```

brandonhewer, 2017-10-05:
Issue #20825
```
SaveIsawPeaks-v1
SaveIsawUB-v1
SaveLauenorm-v1
SaveMD-v1
SaveMD-v2
SaveMDWorkspaceToVTK-v1
SaveMask-v1
SaveNISTDAT-v1
SaveNXSPE-v1
SaveNXTomo-v1
SaveNXcanSAS-v1
SaveNexus-v1
SaveNexusProcessed-v1
SaveOpenGenieAscii-v1
SavePAR-v1
```

brandonhewer, 2017-10-05:
Issue #20827 
```
SavePDFGui-v1
SavePHX-v1
SaveParameterFile-v1
SavePlot1D-v1
SavePlot1DAsJson-v1
SaveRKH-v1
SaveSESANS-v1
SaveSPE-v1
SaveTBL-v1
SaveToSNSHistogramNexus-v1
SaveVulcanGSS-v1
SaveYDA-v1
SaveZODS-v1
Scale-v1
ScaleX-v1
```

josephframsay, 2017-10-05:
```
SelectCellOfType-v1
SelectCellWithForm-v1
SelectNexusFilesByMetadata-v1
SetBeam-v1
SetDetScale-v1
SetGoniometer-v1
SetInstrumentParameter-v1
SetMDFrame-v1
SetSample-v1
SetUB-v1
SetupILLD33Reduction-v1
ShiftLogTime-v1
ShowPeakHKLOffsets-v1
ShowPossibleCells-v1
SignalOverError-v1
SimpleShapeMonteCarloAbsorption-v1
SimulatedDensityOfStates-v1
SingleCrystalDiffuseReduction-v1
SliceMD-v1
SliceMDHisto-v1
```

NickDraper, 2017-10-09:
I've taken 
```
SmoothData-v1
SmoothMD-v1
SmoothNeighbours-v1
SofQW-v1
SofQWCentre-v1
SofQWMoments-v1
SofQWMomentsScan-v1
SofQWNormalisedPolygon-v1
SofQWPolygon-v1
SolidAngle-v1
SortByQVectors-v1
SortDetectors-v1
SortEvents-v1
SortHKL-v1
SortPeaksWorkspace-v1
SortTableWorkspace-v1
SpatialGrouping-v1
SpecularReflectionCalculateTheta-v1
SpecularReflectionCalculateTheta-v2
SpecularReflectionPositionCorrect-v1
SpecularReflectionPositionCorrect-v2
SphericalAbsorption-v1
SplineBackground-v1
SplineInterpolation-v1
SplineSmoothing-v1
Squares-v1
StartLiveData-v1
StatisticsOfPeaksWorkspace-v1
StatisticsOfTableWorkspace-v1
Stitch1D-v3
Stitch1DMany-v1
StringToPng-v1
StripVanadiumPeaks-v1
StripVanadiumPeaks-v2
SuggestTibCNCS-v1
SuggestTibHYSPEC-v1
SumEventsByLogValue-v1
SumNeighbours-v1
SumRowColumn-v1
SumSpectra-v1
SwapWidths-v1
Symmetrise-v1
```

NickDraper, 2017-10-09:
I've taken
```
TransformHKL-v1
TransformToIqt-v1
Transpose-v1
TransposeMD-v1
UnwrapMonitor-v1
UnwrapMonitorsInTOF-v1
UnwrapSNS-v1
UpdateInstrumentFromFile-v1
UpdatePeakParameterTableValue-v1
UpdateScriptRepository-v1
VelocityAutoCorrelations-v1
VelocityCrossCorrelations-v1
VesuvioCalculateGammaBackground-v1
VesuvioCalculateMS-v1
VesuvioCorrections-v1
VesuvioDiffractionReduction-v1
VesuvioL1ThetaResolution-v1
VesuvioPeakPrediction-v1
VesuvioPreFit-v1
VesuvioResolution-v1
VesuvioTOFFit-v1
VesuvioThickness-v1
ViewBOA-v1
VisionReduction-v1
WeightedMean-v1
WeightedMeanOfWorkspace-v1
WienerSmooth-v1
WorkflowAlgorithmRunner-v1
```

NickDraper, 2017-10-09:
Taking the concepts
```
concepts/Algorithm
concepts/CrystalStructureAndReflections
concepts/EventWorkspace
concepts/FABADA
concepts/FacilitiesFile
concepts/Fitting
concepts/HistogramData
concepts/HowToDefineGeometricShape
concepts/Howtoresetdetectorscalibration
concepts/IndexProperty
concepts/Instrument
concepts/InstrumentAccessLayers
concepts/InstrumentDataService
concepts/InstrumentDefinitionFile
concepts/InstrumentParameterFile
concepts/LETSampleIDF
concepts/MDHistoWorkspace
concepts/MDWorkspace
concepts/MatrixWorkspace
concepts/ORNL_SANS_Reduction
concepts/PeaksWorkspace
concepts/PointAndSpaceGroups
concepts/PropertiesFile
concepts/Run
concepts/SANS2DSampleIDF
concepts/Sample
concepts/SampleEnvironment
concepts/SymmetryGroups
concepts/TableWorkspaces
concepts/UnitFactory
concepts/Workspace
concepts/Workspace2D
concepts/WorkspaceGroup
concepts/calibration/CalFile
concepts/calibration/TubeCalibration
```

NickDraper, 2017-10-11:
Taken
Fit Functrions
```
fitfunctions/CrystalFieldHeatCapacity
fitfunctions/CrystalFieldMagnetisation
fitfunctions/CrystalFieldMoment
fitfunctions/CrystalFieldSusceptibility
fitfunctions/ElasticIsoRotDiff
fitfunctions/InelasticIsoRotDiff
fitfunctions/IsoRotDiff
fitfunctions/LatticeFunction
fitfunctions/PrimStretchedExpFT
fitfunctions/StretchedExpFT
fitfunctions/TabulatedFunction
fitfunctions/TeixeiraWaterSQE
```

NickDraper, 2017-10-11:
Once the last of these PR's is done this can be closed

rosswhitfield, 2017-10-11:
We are waiting on @samueljackson92 with #20759 and @edwardb96 with #20715

rosswhitfield, 2017-10-16:
Still waiting on @edwardb96 with #20715

edwardb96, 2017-10-16:
Sorry @rosswhitfield I should have put a note on the issue/unassigned myself.

I wasn't able to get the doc tests to run without crashing (fedora 26 - search xcb on slack tech-qa) so if someone else wants to take on #20715 or knows how to get them to run without producing 

```
X Error: GLXBadContext 170
  Extension:    155 (Uknown extension)
  Minor opcode: 6 (Unknown request)
  Resource id:  0x2e000a9
[xcb] Unknown sequence number while processing queue
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
MantidPlot: xcb_io.c:259: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.
Aborted (core dumped)
```  

that would be great.

PR #20958 (2017-10-19): Fix remaining doc tests for python 3 and enable doc-tests on PR builds
## Description of work.
This is the last PR part of refactoring effort to bring Python3 compatiblity into the doctests.
So this closes #20672 

Merging this will also enable doc-tests on the `Ubuntu Python 3` PR build.

## To test.

 - Code review.

Fixes #20759
Internal change, no release notes
---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform [... cut]
closed 2017-10-19 by SimonHeybrock