mantidproject/mantid #20706: Sample Transmission Calculator - Output graph does not resize for negatives

MatthewBowles, 2017-09-28:
### Expected behaviour

When a wavelength range is entered, the x axis of the output graph should resize to show only this range.

### Actual behaviour

If the wavelengths entered are negative (presumably to indicate direction) then the x axis will not resize properly and will always show zero, even though there is no data between the 'High' value and zero.

### Steps to reproduce the behaviour

-Open Sample Transmission Calculator (interfaces -> general)
-Enter valid chemical formula (e.g. H2-O) and bin width
-Set both Low and High values for the wavelength range as negatives
-Click calculate

### Platforms affected
Seen on Windows 7

SpencerHowells, 2017-09-28:
Wavelengths cannot be negative!!

SpencerHowells, 2017-09-29:
@MatthewBowles Negative wavelengths are a physical impossibilty!! If you want to make any change make it impossible to input negative values.

MatthewBowles, 2017-09-29:
@SpencerHowells Yes, I have already edited this issue to reflect this.

PR #21071 (2017-11-02): Sample Transmission Calculator - Do not allow negative wavelengths in the spin boxes
This is a simple change to set 0 as the minimum value in the wavelength spin boxes in the sample transmission claculator.

**To test:**

1. Open Mantid,
1. Interfaces-> General->Sample Transmission Calculator
1. Try to enter a negative value in the wavelength boxes.

Fixes #20706. 

**Release Notes** 
*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x]  [... cut]
closed 2017-11-02 by martyngigg