mantidproject/mantid #20746: LoadLiveData does not time out

martyngigg, 2017-10-02:
After testing the fix for #20703 it was noticed that `LoadLiveData` never times out if it can't connect to the server. Mantid is left in a state that requires a user to kill it rather than being able to quit cleanly.

NickDraper, 2017-10-24:
Attempted to reproduce with offspec and disabling the network card:

Produced these warnings, but did not crash or hang:
```
error reading command (getdat)
Unable to read DATA from DAE 130.246.37.30
Error in execution of algorithm MonitorLiveData:
Unable to read DATA from DAE  in 130.246.37.30
```

NickDraper, 2017-10-24:
repeated with pulling out the network cable earlier before the first dataset was created. 

Then no crash or hang, but you get an uncancellable algorithm
[image]

NickDraper, 2017-10-24:
Attempt 3:
Pulling the plug just after the first dataset gives a stuck algoirthm again, but let me search through the threads to see what we were stuck on.

line 100 of isisds_command.cpp
``` C++
static int recv_all(SOCKET s, void *buffer, int len, int flags) {
  char *cbuffer = reinterpret_cast<char *>(buffer);
  int n, ntot;
  ntot = 0;
  while (len > 0) {
    n = recv(s, cbuffer, len, flags);   //< THIS LINE IS THE PROBLEM
    if (n <= 0) {
      return n; /* error */
    }
    len -= n;
    cbuffer += n;
    ntot += n;
  }
  return ntot;
}```

NickDraper, 2017-10-24:
After setting a timeout on the socket you get an error message like this:
```
Loading live data chunk 0 at 15:47:18
error receiving command NTC1 (getpar)
Unable to read NTC1 from DAE 130.246.37.30
Error in execution of algorithm MonitorLiveData:
Unable to read NTC1 from DAE  in 130.246.37.30
```
Not perfect, but much better and no hanging algrorithms

NickDraper, 2017-10-24:
This unfortunately times out on long but otherwise successful operations as well as genuine network failures.  Darn

NickDraper, 2017-10-24:
so a default timeout of 120 sec, but it should attempt to read it from the config service (just don't put it in the file by default)

PR #21023 (2018-07-02): Load isis live timeout for histogram data
This has added a timeout to the ISIS Histogram DAE listener.  Unfortunately the timeout cannot seperate network failure from long running successful operations, so the value of the timeout has been set to 120 seconds by default.

You can configure it in the user.properties file with the key:
```
ISISDAE.Timeout = 20 #seconds
```

**To test:**

**NOTE: This has some windows/linux specific code, it has been developed on windows so I'm pretty confident about it working their, but it needs testing on Linux/Mac.**

1. Run Mantid, Load->Live Data
1. Pick an  [... cut]
closed 2018-07-02 by ianbush