mantidproject/mantid #20757: Indirect - IndirectSampleChanger/ElasticWindowMultiple - %d format Receives String

brandonhewer, 2017-10-03:
In the execution of IndirectSampleChanger, a bug was found in which it is possible to receive the following error:

`
Error in execution of algorithm EnergyWindowScan:
%d format: a number is required, not str
at line 152 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\ElasticWindowMultiple.py'
caused by line 331 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\ElasticWindowMultiple.py'
at line 172 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\EnergyWindowScan.py'
Error in execution of algorithm IndirectSampleChanger:
%d format: a number is required, not str
at line 152 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\ElasticWindowMultiple.py'
caused by line 331 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\ElasticWindowMultiple.py'
at line 172 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\EnergyWindowScan.py'
at line 103 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\IndirectSampleChanger.py'
`

SpencerHowells, 2017-10-03:
IRIS now uses IBEX to run the Sample Environment. The output into the Sample Logs needs to be sorted out first. The we look at putting the parameter names into the IPF.

brandonhewer, 2018-03-05:
Solved in https://github.com/mantidproject/mantid/pull/21784

closed 2018-03-05 by brandonhewer