mantidproject/mantid #20832: String constraints validator documentation

AndreiSavici, 2017-10-05:
In PR #20368, StringContainsValidator was exposed to python. However, it lacks a documentation page :(
In the release notes, there is already an entry for the validator, just need to add a link to the documentation page.

Should be part of 3.11 release

martyngigg, 2017-10-06:
Good spot, Andrei. The builds are already done for 3.11 but it will at least get to the web documentation when it's put in master.

@KarlPalmen Can you have a look at this, please?

KarlPalmen, 2017-10-06:
The only documentation I can find about _any_ validator is in the python course https://www.mantidproject.org/Extending_Mantid_With_Python which describes workspace validators at
https://www.mantidproject.org/Workspace_Validators and combining validators at
https://www.mantidproject.org/Combining_Validators. 

I therefore feel that this issue should from part of a project in improve documentation about _all_ validators, if we want this.

martyngigg, 2017-10-06:
@KarlPalmen Andrei is referring to the pages at http://docs.mantidproject.org/nightly/api/python/mantid/api/index.html

As we have to fix a problem with the release notes links I will add the skeleton file for this too.

KarlPalmen, 2017-10-06:
Found document source in ...\docs\source\api\python\mantid\api\ .

KarlPalmen, 2017-10-06:
When I was working on  issue #19010 , I was unaware of http://docs.mantidproject.org/nightly/api/python/mantid/api/index.html and when I asked @mantid-roman where to put the documentation, he advised ``Concepts``, so I put it there. 
I found a search facility at the Python API http://docs.mantidproject.org/nightly/api/python/index.html and cannot find that documentation, when I search ``fitFunctions``.

KarlPalmen, 2017-10-06:
A further thought is that http://docs.mantidproject.org/nightly/api/python/mantid/api/index.html is only for the C++ interface and not for any wrappers.

PR #20843 (2017-10-06): Add page for StringContainsValidator
**Description of work**

Adds a page for `StringContainsValidator`.

**To test:**

Build the docs and check that the page for `StringContainsValidator` exists. 

Fixes #20832 

**Release Notes** 

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the  [... cut]
closed 2017-10-07 by AndreiSavici