mantidproject/mantid #20836: Engineering diffraction

NickDraper, 2017-10-06:
Gui freezes Mantid and give useless error dialog

[image]

AntonPiccardoSelg, 2017-10-06:
The dialog appears by design. 
The issue with freezing is related to the `MWRunFiles` widget [here](https://github.com/mantidproject/mantid/blob/master/qt/scientific_interfaces/EnggDiffraction/EnggDiffractionViewQtGUI.cpp#L136:L137) which then triggers the `MWRunFiles::findFile` method which then spends a lot of time in [here](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/src/MWRunFiles.cpp#L707).

It is not clear how we could improve this at the moment.

NickDraper, 2017-10-06:
If you turn off searching the data archive the GUI starts instantly.

I'm looking at if you can turn off searching for the default files numbers when you first start up.

PR #20847 (2017-10-06): Fix for Mantid freezing when handling Engineering Diffraction
Fixes #20836

The fixes stop the GUI freezing up (However quitting the thread is still taking ages. Maybe somebody should look into this). 

**To test:**

* Open MantidPlot
* Open Interfaces > Diffraction > Engineerig Diffraction
  * Confirm that it does not freeze up Mantid and starts quickly
* Close the interface
  * Confirm that not freeze up Mantid when shutting down
* Close the Mantid
  * Confirm that there are not error messages

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproj [... cut]
closed 2017-10-06 by NickDraper