mantidproject/mantid #20865: Fit benchmarking adding Muon data

AnthonyLim23, 2017-10-09:
This is to allow the fit benchmarking tool to fit to muon data (outside of neutron file). @Anders-Markvardsen is there a good place to update the fitting data (i.e. add the new muon data)?

AnthonyLim23, 2017-10-11:
@Anders-Markvardsen can you run the neutron data in fit benchmarking?

Anders-Markvardsen, 2017-10-19:
Hi @AnthonyLim23 to answer your question above when I run the fit benchmarking with 'neutron' I get the error:

"Saved neutron_enginx 193749 calibration, spectrum 651, peak 19.txt to C:\Backup\Backup_folder1\work\code\Mantid\builds\all\MantidPlot
Saved neutron_enginx 193749 calibration, spectrum 651, peak 19.html to C:\Backup\Backup_folder1\work\code\Mantid\builds\all\MantidPlot
RuntimeError: Cannot create graph, see log for details.
  at line 62 in 'C:/Backup/Backup_folder1/work/code/Mantid/fitting/FitMinimizerTestingAnthony/runScripts.py'
  caused by line 74 in 'C:/Backup/Backup_folder1/work/code/Mantid/git/mantid/scripts/CompareFitMinimizers\results_output.py'
  caused by line 188 in 'C:/Backup/Backup_folder1/work/code/Mantid/git/mantid/scripts/CompareFitMinimizers\results_output.py'
  caused by line 231 in 'C:/Backup/Backup_folder1/work/code/Mantid/git/mantid/scripts/CompareFitMinimizers\results_output.py'
  caused by line 38 in 'C:\Backup\Backup_folder1\work\code\Mantid\builds\all\bin\Release\pymantidplot\qtiplot.py'
  caused by line 276 in 'C:\Backup\Backup_folder1\work\code\Mantid\builds\all\bin\Release\pymantidplot\__init__.py'"

More specifically Mantid fails when it tried to plot 'ENGINX 193749 calibration, spectrum 651, peak 20. I am running it here with the same two minimizers as done when testing the pull request. This failure seems to be to be unrelated to code changes you have made in this pull request, but is picking a bug? which has been introduced into mantid. Could you look into this please? If unrelated to this issue then I suggest this is fixed in a separate issue.   

Below is zip file containing the workspace created which looks a bit strange (so maybe something changed with fiting perhaps) and which is offending Mantid when you try to plot it (so maybe something changed there) 

[peak20.zip](https://github.com/mantidproject/mantid/files/1399296/peak20.zip)

Anders-Markvardsen, 2017-10-20:
Just ran it again on a clean build, the same error: Unexpected exception: bad allocation , although possibly at a slightly different location. My log for this test was:

"Parameters:
  Name: OutputWorkspace, Value: ENGINX 193749 cal ... trum 651, peak 19, Default?: No, Direction: Output
  Name: DataX, Value: 3.87869e-315,-9.6 ... 75835e-315,0,-nan, Default?: No, Direction: Input
  Name: DataY, Value: 2.70236e-315,-1.2 ... -315,2.28303e-317, Default?: No, Direction: Input
  Name: DataE, Value: , Default?: Yes, Direction: Input
  Name: NSpec, Value: 1, Default?: Yes, Direction: Input
  Name: UnitX, Value: , Default?: Yes, Direction: Input
  Name: VerticalAxisUnit, Value: SpectraNumber, Default?: Yes, Direction: Input
  Name: VerticalAxisValues, Value: , Default?: Yes, Direction: Input
  Name: Distribution, Value: 0, Default?: Yes, Direction: Input
  Name: YUnitLabel, Value: , Default?: Yes, Direction: Input
  Name: WorkspaceTitle, Value: , Default?: Yes, Direction: Input
  Name: ParentWorkspace, Value: , Default?: Yes, Direction: Input
  Name: ParallelStorageMode, Value: Parallel::StorageMode::Cloned, Default?: Yes, Direction: Input
CreateWorkspace successful, Duration 0.01 seconds
Unexpected exception: bad allocation
Unexpected exception: bad allocation"

AnthonyLim23, 2017-10-20:
@Anders-Markvardsen  did you try this on the new version (from the PR)? I have just tried it and it seems to be going along happily.

Anders-Markvardsen, 2017-10-20:
uumm yes - could you pass by and check what I see ~9:15

AnthonyLim23, 2017-10-20:
ok

AnthonyLim23, 2017-10-20:
I think it is the name change that killed it. Just looked at my machine and it had completed without any errors. 

[image]

Anders-Markvardsen, 2017-10-20:
good news:) Yes you were right. By taking out 

    group_names = ['neutron']
    group_suffix_names = ['neutron']

above line 39 it ran OK. 

To reproduce the problem I see. After executing the full script run:

group_names = ['NIST, "lower" difficulty']
group_suffix_names = ['nist_lower']
    
for idx, group_results in enumerate(results_per_group):
        print("\n\n")
        print("********************************************************")
        print("**************** RESULTS FOR GROUP {0}, {1} ************".format(idx+1,
                                                                                group_names[idx]))
        print("********************************************************")
        fitout.print_group_results_tables(minimizers, group_results, problems[idx],
                                          group_name=group_suffix_names[idx],
                                          use_errors=use_errors,
                                          simple_text=True, rst=True, save_to_file=True, color_scale=color_scale)

which works but then change the first two lines to e.g.

group_names = ['Neutron data']
group_suffix_names = ['neutron_data']

and it fails - do you see this?

AnthonyLim23, 2017-10-20:
I will try this over lunch. If it breaks I will open a new issue

AnthonyLim23, 2017-10-20:
the following seemed to work 

```
#group_names = ['NIST, "lower" difficulty', 'NIST, "average" difficulty', 'NIST, "higher" difficulty', "CUTEst", "Neutron data"]
group_names = [ "Neutron data"]
#group_suffix_names = ['nist_lower', 'nist_average', 'nist_higher', 'cutest', 'neutron_data']
group_suffix_names = [ 'neutron_data']
```

Anders-Markvardsen, 2017-10-20:
Umm this doesn't work for me. I will separately look into why my pc is unhappy about this

AnthonyLim23, 2017-10-20:
huh, strange

AnthonyLim23, 2017-10-20:
huh, strange

PR #20907 (2017-11-03): Running muon and neutron data in fit benchmarking
The fit bench marking should be able to use neutron and muon data. This required adding the full path to the load algorithm and making muon an acceptable argument.

There are few other changes to allow fit bench marking work with the new output from the fit algorithm. 

**To test:**

Download the attached zip file: 

[FitMinimizerTesting.zip](https://github.com/mantidproject/mantid/files/1379578/FitMinimizerTesting.zip)

Unzip it

In the python scripting window open runScripts.py

Then on line 35 (run_data) you can choose what to run (muon, neu [... cut]
closed 2017-11-03 by ianbush