mantidproject/mantid #20888: Interface documentation offline help

AnthonyLim23, 2017-10-10:
all of the interface help icons should redirect to the offline help and at present some go to online help.

martyngigg, 2017-10-10:
We need to be careful here as things like the ISIS SANS v1 interface doesn't have any offline help and as it's going to be replaced we may not put the work into transferring it from the wiki.

The others should be done though. Could you put a list of affected interfaces here, please?

AnthonyLim23, 2017-10-10:
the following all redirect to online help:

convert to energy transfer 
step scan interface (in general)
scd data reduction (page is empty)

felt like more

AnthonyLim23, 2017-10-12:
These pages are online only. So I am closing this issue.

closed 2017-10-12 by AnthonyLim23