mantidproject/mantid #20889: HFIR Powder Diffraction

AnthonyLim23, 2017-10-10:
when you open HFIR Powder diffraction reduction a warning about the file location is generated. 

But, it then opens ok. When I try to load data I get an error message 

```
File "C:/Users/BTR75544/MantidInstall/scripts\HFIRPowderReduction\HfirPDReductionGUI.py", line 650, in doLoadData
    status, datafilename = self._uiDownloadDataFile(exp=expno, scan=scanno)
  File "C:/Users/BTR75544/MantidInstall/scripts\HFIRPowderReduction\HfirPDReductionGUI.py", line 2143, in _uiDownloadDataFile
    status, errmsg = urllib.downloadFile(fullurl, srcFileName)
AttributeError: 'module' object has no attribute 'downloadFile
```

### Expected behavior

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

AnthonyLim23, 2017-10-10:
[image]

peterfpeterson, 2017-10-10:
There appear to be two issues
1. The warning dialog looks scary, but probably is just informational
2. The missing function on `urllib`

For the latter, I cannot easily test it, but it looks like the offending call should be replaced with `urllib.urlretrieve(fullurl, srcFileName)`

PR #20896 (2017-10-11): Fix file downloading issue with HFIR PD GUI 
Fixed the wrong import of urllib in HFIR Powder Diffraction interface.

**To test:**

Try Exp 596 Scan 11.  File shall be downloaded.

Fixes #20889. 

**Release Notes** 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class in isolation?
- [ ] If ther [... cut]
closed 2017-10-11 by AnthonyLim23