mantidproject/mantid #20926: MantidPlot cannot load POLARIS NeXus files.

edwardb96, 2017-10-16:
Attempting to load `POLARIS00098534.nxs` from the ISIS Sample Data does not work.
- The load begins, and makes no progress.
- Trying to close MantidPlot results in it being reported as not responding.

### Expected behavior
The load would succeed.

### Actual behavior
The load makes no progress.

### Steps to reproduce the behavior
1. Go to Workspaces->Load
2. Select POLARIS00098534.nxs from the ISIS sample data.
3. Nothing...

### Platforms affected
Fedora 26, Ubuntu 16.04 - Mantid 3.11+

NickDraper, 2017-10-23:
Can confirm this on windows with Mantid 3.11, a more recent file seems to work though

NickDraper, 2017-10-23:
However this just worked for me with the current master!

closed 2017-10-23 by edwardb96