mantidproject/mantid #20943: Data Loader for FDA

AnthonyLim23, 2017-10-17:
Create a widget for loading/selecting data in the FDA GUI. It will include 

- A load current run button
- A forward button
- A backward button
- A search bar (can type run numbers in)
- The group/pair to plot

a temporary home tab will also be made which will include instrument selection. This will be extended at a later date.

TWJubb, 2018-07-30:
[LoadWidget.pdf](https://github.com/mantidproject/mantid/files/2240804/LoadWidget.pdf)

Initial sketch of how the LoadWidget should look.

The top row handles file names, the middle row handles run numbers and the bottom row describes the behaviour when multiple files are loaded at once.

- The **Browse** button opens a file browser dialog showing only `.nxs` files. It should allow one or more files to be selected, it will then load the workspaces and add a delimeter separated list of fully-qualified file paths to the top line edit, the "C" button (to be replaced with an appropriate icon) will cause the entire list to be copied to the clipboard so the user can use it elsewhere if needed.

- The **Load Current Run** button uses the currently selected instrument (which is also conveniently displayed next to the button, to find the latest run, loads it, and puts the run number in the middle line edit.

- The middle line edit is also filled when loading from the browse button. The user can input run numbers using `,` as a separator, and `-` as a range specifier. These will then be loaded after hitting enter.

- Finally the bottom combo box allows the users to change the behaviour of the loading when more than one run is requested. They can choose either "Co-Add", i.e sum all the runs into one workspace, or "Simultaneous" where each run will be loaded as a separate workspace.

AnthonyLim23, 2018-07-30:
we will also need the option to hide the combo box.

closed 2019-02-11 by AnthonyLim23