mantidproject/mantid #20961: MantidPlotMatplotlibTest fails with matplotlib v2.1.0

rosswhitfield, 2017-10-18:
MantidPlotMatplotlibTest passed with `matplotlib v2.0.2` but fails with [`v2.1.0`](https://github.com/matplotlib/matplotlib/releases/tag/v2.1.0).

```
1698: Test command: /home/rwp/mantid/Release/master/bin/MantidPlot "-xq" "/home/rwp/mantid/MantidPlot/test/MantidPlotMatplotlibTest.py"
1698: Environment variables: 
1698:  MANTID_SOURCE=/home/rwp/mantid
1698:  PYTHONPATH=/home/rwp/mantid/Release/master/python-xmlrunner-src
1698:  MANTID_SCREENSHOT_REPORT=/home/rwp/mantid/Release/master/bin/screenshots
1698: Test timeout computed to be: 300
1698: ConfigService-[Notice] MessageDisplayPriority log channel set to PRIO_NOTICE priority
1698: FrameworkManager-[Notice] Welcome to Mantid 3.11.20171018.1431
1698: FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
1698: 
1698: Running tests...
1698: ----------------------------------------------------------------------
1698: EE
1698: ======================================================================
1698: ERROR [0.201s]: test_1d_plot (__main__.MantidPlotMatplotlibTest)
1698: ----------------------------------------------------------------------
1698: Traceback (most recent call last):
1698:   File "/home/rwp/mantid/MantidPlot/test/MantidPlotMatplotlibTest.py", line 31, in test_1d_plot
1698:     ax = plt.plot(x,y)
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 3229, in plot
1698:     ax = gca()
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 959, in gca
1698:     return gcf().gca(**kwargs)
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 588, in gcf
1698:     return figure()
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 534, in figure
1698:     **kwargs)
1698:   File "/home/rwp/mantid/Release/master/bin/pymantidplot/mpl/backend_mtdqt4agg.py", line 67, in __call__
1698:     six.reraise(*self._exc_info)
1698:   File "/home/rwp/mantid/Release/master/bin/pymantidplot/mpl/backend_mtdqt4agg.py", line 77, in on_call
1698:     self.callee(*self._args, **self._kwargs)
1698:   File "/home/rwp/mantid/Release/master/bin/pymantidplot/mpl/backend_mtdqt4agg.py", line 111, in _new_figure_manager_impl
1698:     figure_class = kwargs.pop('FigureClass', Figure)
1698: NameError: global name 'Figure' is not defined
1698: 
1698: ======================================================================
1698: ERROR [0.212s]: test_image_plot (__main__.MantidPlotMatplotlibTest)
1698: ----------------------------------------------------------------------
1698: Traceback (most recent call last):
1698:   File "/home/rwp/mantid/MantidPlot/test/MantidPlotMatplotlibTest.py", line 47, in test_image_plot
1698:     vmax=abs(Z).max(), vmin=-abs(Z).max())
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 3064, in imshow
1698:     ax = gca()
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 959, in gca
1698:     return gcf().gca(**kwargs)
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 588, in gcf
1698:     return figure()
1698:   File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 534, in figure
1698:     **kwargs)
1698:   File "/home/rwp/mantid/Release/master/bin/pymantidplot/mpl/backend_mtdqt4agg.py", line 67, in __call__
1698:     six.reraise(*self._exc_info)
1698:   File "/home/rwp/mantid/Release/master/bin/pymantidplot/mpl/backend_mtdqt4agg.py", line 77, in on_call
1698:     self.callee(*self._args, **self._kwargs)
1698:   File "/home/rwp/mantid/Release/master/bin/pymantidplot/mpl/backend_mtdqt4agg.py", line 111, in _new_figure_manager_impl
1698:     figure_class = kwargs.pop('FigureClass', Figure)
1698: NameError: global name 'Figure' is not defined
1698: 
1698: ----------------------------------------------------------------------
1698: Ran 2 tests in 0.413s
1698: 
1698: FAILED (errors=2)
1698: 
1698: Generating XML reports...
1698: ApplicationWindow-[Fatal] Fatal error on line 54 of "/home/rwp/mantid/MantidPlot/test/MantidPlotMatplotlibTest.py" encountered:
1698: ApplicationWindow-[Fatal] SystemExit: 1
1698: ApplicationWindow-[Fatal]   at line 54 in '/home/rwp/mantid/MantidPlot/test/MantidPlotMatplotlibTest.py'
1698: ApplicationWindow-[Fatal]   caused by line 64 in '/home/rwp/mantid/MantidPlot/test/mantidplottests.py'
1698: MantidUI-[Notice] MantidPlot is shutting down...
1/1 Test #1698: MantidPlotMatplotlibTest.py ......***Failed    8.03 sec
```

@martyngigg Any thoughts?

rosswhitfield, 2017-10-18:
Looks like [`from matplotlib.backends.backend_qt4agg import *`](
https://github.com/mantidproject/mantid/blob/master/MantidPlot/pymantidplot/mpl/backend_mtdqt4agg.py#L21) no longer imports `Figure`, I think it was matplotlib/matplotlib#8773

rosswhitfield, 2017-10-18:
This fixes it, unless these is a better solution?

```diff
--- a/MantidPlot/pymantidplot/mpl/backend_mtdqt4agg.py
+++ b/MantidPlot/pymantidplot/mpl/backend_mtdqt4agg.py
@@ -19,6 +19,7 @@ except ImportError:
     from matplotlib.backends.qt4_compat import QtGui as QtWidgets
 # Import everything from the *real* matplotlib backend
 from matplotlib.backends.backend_qt4agg import *
+from matplotlib.figure import Figure
 import six
 
 # Remove the implementations of new_figure_manager_*. We replace them below
```

AndreiSavici, 2017-10-18:
I can run `from matplotlib.figure import Figure` in matplotlib 1.5.3 (what we have on RHEL machines at ORNL)

martyngigg, 2017-10-20:
@rosswhitfield For backwards compatability I think your idea seems like the best solution.

martyngigg, 2017-10-20:
I'd really like to get away from requiring our own backend for the workbench but the only way I can see to do it is a multi-process model, which comes with its own complications.

PR #21003 (2017-10-23): Fix mpl backend for matplotlib >= 2.1.0
See issues #20961 for details

Fixes #20961

**Release Notes** 
*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class in isolation?
- [ ] If there are changes in the release notes then do they describe the change [... cut]
closed 2017-10-23 by AndreiSavici