mantidproject/mantid #21062: Unscripted Testing Live Data ISIS

Mantid-Matthew, 2017-10-27:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than an hour on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* Talk to the developer if possible
* create an issue if it is new against the current release

gemmaguest, 2017-10-27:
Testing on Windows 7

gemmaguest, 2017-10-30:
I left 2 MantidPlots both with multiple MonitorLiveData algorithms running over the weekend, one using ISIS_Event with FakeISISEventDAE and one using a real instrument (Offspec), using the settings from #20717. I had slice viewer open for the latter. Both were still running ok on Monday morning, although:

- When I tried to cancel the monitor algorithm for Offspec, it did not finish, and stayed saying "Cancelling". I tried cancelling all of the other monitor algorithms (I had about 6) and got the same thing. Mantid then hung and I had to kill it. This is probably related to #19040.
- MantidPlot was using around 1.7GB for the instance using the fake instrument, and 1.2GB for the Offspec instance. Even after cancelling all algorithms and deleting all workspaces the former was still using 1.2GB. Previously when testing #20717 on Windows 10, Mantid was using only around 300MB.

closed 2018-07-06 by Mantid-Matthew