mantidproject/mantid #21157: CalculatePaalmanPings GUI crash

gvardany, 2017-11-09:
The GUI tab crashes if a file containing a workspace group is supplied.

SpencerHowells, 2017-11-09:
Why would you want to run on a group? What varies for the components? Sample or temperature or what? What would the program do, scientifically, for the individual components?

gvardany, 2017-11-09:
We agreed earlier that it should not run for a group. The issue is just to prevent the crash with a nicer error popping up. The different items in the group are for different doppler energies in our case.

SpencerHowells, 2017-11-09:
Just checking. We already have it as Issue #19209 & I’d asked Brandon to give a message that groups cannot de used.

gvardany, 2017-11-09:
I have forgotten about that one. It seems the same, but it can be that the same problem exists both in the algorithm and the GUI tab layer, one should test both at the end.

SpencerHowells, 2017-11-09:
Good point. We'll do both as suggested.

gvardany, 2017-12-12:
@SpencerHowells What is your timescale to work on this? Or should I attack this now? This is reported by a user, so it would be nice to treat it timely.

SpencerHowells, 2017-12-12:
@brandonhewer What is the answer for Gagik?

brandonhewer, 2017-12-12:
@gvardany Indeed, the necessary checks were supposed to be added in the original PR:
https://github.com/mantidproject/mantid/pull/20373

I don't mind taking a look at this immediately, I imagine the timescale will only be a couple of days at most.

gvardany, 2017-12-12:
Thanks for agreeing to work on this. I would suggest to walk through all the indirect monte carlo and paalman pings algorithms and GUI tabs, and harmonise their behaviour for workspace groups. The algorithms can respond with an error explaining the reason in validation step.

closed 2018-01-02 by martyngigg