mantidproject/mantid #21167: Reflectometry GUI fixes

gemmaguest, 2017-11-10:
This is an umbrella issue to keep track of fixes required for the new reflectometry GUI

- [ ] Transmission runs entered on the Settings tab are not found #21076
- [ ] You should be able to specify the transmission run on the settings tab or runs tab as a run number or a workspace name in the ADS - check that this still works.
- [ ] There seems to be a recent regression where no parameters from the GUI are getting passed through to CreateTransmissionWorkspaceAuto
- [ ] Improve output workspace names, including time slices #21004
- [ ] The output data should be automatically saved to ASCII. You should be prompted for the save directory if it is not set on the ASCII tab or if the path is not writable. There should be a tickbox to disable auto-saving (primarily for use when testing where the prompt may become annoying). Base default path for the prompt on the general save directory from the user directories dialog?
- [ ] Remove the DirectBeam box from the GUI and the underlying `RegionOfDirectBeam` property from `ReflectometryReductionOneAuto`. Scientists at ISIS do not use this and don't think it would work well anyway.
- [ ] The `Get Defaults` button on the Settings tab doesn't load some properties from the parameter file, e.g. `DetectorCorrectionType`. I think there may be something we need to enable in the code for this. Scientists would like to be able to set all of them from the parameter file.
- [ ] Expose the Q-summation functionality on the Settings tab (`SummationType` and `ReductionType` drop downs).
- [ ] Expose the `CorrectDetectors` tick box on the Settings tab.
- [ ] Polarisation corrections should be handled as a set - currently there are 4 separate boxes and it's not clear they belong together (you would generally edit all or none). This might be better as a single text box with 4 lines in it (containing 4 to 7 numbers per line). Longer term (not a priority for now) Jos would like to be able to load them from a nexus file which could be created on ICat or local directory in mantid's path on the fly and found by the gui just by specifying its name. Ideally it would then also populate the text box which could then be edited by hand.
- [ ] We should make sure all elements have tooltips or What's This help available where possible (preferably the former).
- [ ] We should have a help button link to the wiki.

gemmaguest, 2017-11-23:
@edwardb96 We'll need to coordinate who does what on this list. I'm going to start by looking at the transmission runs stuff.

edwardb96, 2017-11-23:
Is "a recent regression where no parameters from the GUI are getting passed through to CreateTransmissionWorkspaceAuto" fixed by https://github.com/mantidproject/mantid/pull/21237?

gemmaguest, 2017-11-24:
Yes, I think so: #21237 fixes an error about "FirstTransmissionRun not found" and the parameters seem to be getting passed through ok to CreateTransmissionWorkspaceAuto.

gemmaguest, 2018-04-03:
Changed to med priority because the high priority fixes are now complete.

gemmaguest, 2018-06-18:
Cleared milestone because most fixes in here are not high priority

gemmaguest, 2019-05-14:
The slicing and async issues are resolved as part of the restructuring in #23051. The stitching error with different numbers of slices is probably expected.

gemmaguest, 2019-07-23:
Closing because the remaining issues are no longer relevant on the restructured GUI

closed 2019-07-23 by gemmaguest