mantidproject/mantid #21216: Make GDPP process runs on a single child thread

gemmaguest, 2017-11-20:
Currently, the `GenericDataProcessorPresenter` processes each item in the `ISIS Reflectometry` GUI on a separate child thread. The threading was added to avoid locking the whole GUI during processing. However, items are still processed sequentially - it waits for one thread to finish before starting the next item.

Some [queues](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/inc/MantidQtWidgets/Common/DataProcessorUI/GenericDataProcessorPresenter.h#L41) have been added to the `GenericDataProcessorPresenter` to keep track of which item is being processed. We should be able to simplify this by running all processing on a single child thread so that we can remove the queues from the `GenericDataProcessorPresenter`. I think we should also be able to remove [this state flag](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/inc/MantidQtWidgets/Common/DataProcessorUI/GenericDataProcessorPresenter.h#L248)

Note that in the long term we may want to be able to process different groups in parallel. However, this is out of scope for now and the `GenericDataProcessorPresenter` requires quite a bit of tidying up before we can do that in a good way, so for now I think we should just try to simplify things as much as possible.

It may be worth looking at the code before PR #19581, which is when the queues were added.

edwardb96, 2017-11-30:
This does not seem to be as simple as it would sound. It is hard to do this incrementally since: 
1. It involves converting the explicit state machine (consisting of `m_nextActionFlag` and the switch case in `doNextAction`) into an implicit one described by normal control statements. Since this state machine is not cleanly separated from the rest of the code, there are a lot of other potentially conflicting side-effects to contend with/keep track of.
2. The transition from each row being reduced on its own thread to all rows getting reduced on a single thread requires that first we must remove the threading to give a linear code path and then re-add the threading to execute said linear code path. All the while keeping the pause/resume functionality working.

It would probably be easier to revert to #19581 and re-integrate the changes which have happened since then or to rip out the reduction code from the GDDP and start again. Both of these carry significant risk.

gemmaguest, 2017-11-30:
Thanks for looking into this. In that case, we should leave this for now until we have time to do a larger refactoring project in the future.

gemmaguest, 2019-05-14:
The reflectometry GUI is moving away from using the data processor so this is no longer necessary

closed 2019-05-14 by gemmaguest