mantidproject/mantid #21223: ISIS SANS V2 allow for the loading of processed data files.

Mantid-Matthew, 2017-11-20:
Allow for processed Mantid files to be used for the reduction input. Currently we only support nexus from the machine (single and multi period), raw from the machine and added data. Diego would like to process some data and then pass it to the reduction. The main locations where changes are required are load_data.py and file_information.py. It is important that scatter sans data is loaded in a way that the data and the monitors are loaded into separate files. If this can be achieved easily, then this should not be a problem. I assume the main thing is to have a check if the loaded data contains both data and monitors.

Mantid-Matthew, 2018-05-08:
This issue currently seems to be too ill defined to implement. The SANS back end requires that monitors exist and are loaded separately which cannot be guaranteed for an arbitrary processed file. I need to go talk to diego about whether this is still needed and if so what the format of the processed files he requires is.

Mantid-Matthew, 2018-05-24:
Having talked to Diego the processing he is doing is to do with getting the polarization dependent scattering cross sections from the Intensities which are a mixture of them. In the simplest possible terms this boils down to converting the TOF workspace to wavelength, doing some multiplications, converting back to TOF. So this is the type of processed data which needs to be loadable.

RichardHeenan, 2020-07-14:
Suspect this sis the same issue we have with LoKI detector test data from Larmor. We have to use the old (deprecated) SANS interface to read & reduce specially processed nxs files. We may lose the ability top do this in future.

smk78, 2020-07-16:
Also see https://github.com/mantidproject/mantid/issues/25539

martyngigg, 2021-04-01:
I'm just removing the MantidPlot label from the remaining issues that have it to be clear it's not related to that.

rbauststfc, 2023-04-12:
Re-opening as still on to-do list

rbauststfc, 2023-08-14:
I think issue #35875 that I opened recently is actually a duplicate of this one?