mantidproject/mantid #21268: ISIS Powder - Add beam parameter definition

ElliotAOram, 2017-11-27:
**New feature**
**This issue will cover part of #20892**

Its should be possible for users to define the beam parameters (shape of the beam) as this may change and is required for later analysis steps.
The proposed way to do this is to add is as an option to the abstract instrument. This information can be stored as a map:
```
beam_parameters = {width: w, 
                   height: h}
```
At present only rectangle and square beam shapes are used but this could be further updated later as required.

ElliotAOram, 2017-11-27:
Additionally in investigating this issue. set_sample_details has almost the same implementation in `hrpd.py` `polaris.py` and`gem.py` it should be possible to move this functionality to the `abstract_inst.py` parent to avoid code duplication.

SpencerHowells, 2017-11-27:
We’ve put the beam dimensions into the IPF. It can then be the default value in interfaces such as absorption corrections.

ElliotAOram, 2017-11-27:
Good point Spencer, although I don't think the powder instruments use any IPFs currently. There are similar files in place however that maybe (in order to standardise things across techniques) could be candidates for becoming IPFs... I'll make a note to look into this (probably once we start looking into a GUI)

PR #21366 (2017-12-07): isis_powder - add beam parameters
Added the ability to set beam height and width in the isis_powder instruments. This is not yet being used, but will be used for total scattering corrections and may need to be taken into account for other isis_powder instruments at a later stage. Documentation has been updated.

**To test:**
* Run the below script and check the output of `print(polaris._beam_parameters)` to ensure the beam parameters are updated correctly. Try to force errors with different input (negative numbers, strings, etc.)
* Ensure the unit new tests pass locally and on the build server [... cut]
closed 2017-12-07 by martyngigg