mantidproject/mantid #21281: Export base algorithm dialog  & custom dialogs to mantidqt

martyngigg, 2017-11-28:
Following on from #21180. After selecting the algorithm the user will run want to run it. This will require access to the algorithm dialogs, which in turn will require defining the code for executing an algorithm in the background.

Tasks:
* Export `AlgorithmDialog`, `GenericDialog`, `InterfaceManager` (plus related classes) to Python inside `mantidqt`. See #21251 
* Define functionality to run the algorithms asynchronously in the background of the workbench. My suggestion would be that we create a module `workbench.external.mantid` and add this functionality there.

mantid-roman, 2017-11-30:
@martyngigg Should I wait until #21251 is merged?

martyngigg, 2017-11-30:
@mantid-roman I think you should be okay to base the branch of that one. I'm nearly done with it.