mantidproject/mantid #21296: Problems with MSD tab of IDA when using the output from ELWIN with BASIS files

jmborr, 2017-11-29:
### Actual behavior
1 Fitting range not identified:
[image]
2 Incorrect fitting:
[image]

### Expected behavior
Analysis of the same data in IDA of Mantid 3.10 results in a reasonable fit:
[image]

### Steps to reproduce the behavior
Use file 
[BASIS-BASIS_63652_sqw_elwin_eq2.nxs.zip](https://github.com/mantidproject/mantid/files/1514740/BASIS-BASIS_63652_sqw_elwin_eq2.nxs.zip) in IDA's MSD tab with Gaussian model.

### Platforms affected
All

SpencerHowells, 2017-11-29:
@jmborr Which version are you using? Brandon is doing a big re-write of the fitting interfaces issue #21123   harmonising the 4 fitting tabs. They will all use the same graphics methods. We also identified many bugs in 3.11 which will also be sorted out. MsdFit now requires the _eq file not _eq2 in order to fit the 3 models.

jmborr, 2017-11-29:
@SpencerHowells   @brandonhewer  Ouch! I was using `_eq2`, as customary. I tried with `_eq` and I get reasonable fits :smile:.
The [documentation](http://docs.mantidproject.org/nightly/interfaces/Indirect%20Data%20Analysis.html#msd-fit) has not been updated because it reads that the input file in MSD tab is "A file that has been created using the Elwin tab with an x axis of Q^2".

brandonhewer, 2017-11-30:
@jmborr @SpencerHowells I'll look into updating the documentation 😄

PR #21324 (2017-12-05): Problems with MSD tab of IDA when using the output from ELWIN with BASIS files
Wrote a system test.

**To test:** it's a system test, so only code review is necessary
Fixes #21296
*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class in isolation?
- [  [... cut]
closed 2017-12-05 by OwenArnold